nanopb/nanopb #621: skip_message option for submessages ?

geoffreymbrown, 2020-12-04:
The following is for proto3.  Suppose I have a message type foo and another message bar

message bar { 
    int32 a = 1;
    foo   b = 2;
}

and in the appropriate options file I have

foo skip_message:true

while the structure definition for foo is deleted,  the structure definition for bar still contains a field of type foo.  

I guess my expectation is that this field would have been left out.

PetteriAimonen, 2020-12-04:
Yeah, it could make sense for the dependent fields to be skipped also.

For now, you can add `bar.b type:FT_IGNORE`

PetteriAimonen, 2024-09-03:
I'm closing off this old issue, the workaround with FT_IGNORE seems good enough. Automatically removing dependent fields wouldn't work reliably across multiple files.

closed 2024-09-03 by PetteriAimonen