nanopb/nanopb #623: Easier out of the box handling of repeated and nested structs.

baryluk, 2020-12-15:
Hi,

I was wondering, I want to use protocol buffers in C on desktop Linux, (for unrelated reasons we can't use full C++ unfortunately), and nanopb works pretty nice. However, the fact that we have a lot of message types in our projects, repeated fields, and some nested stuff (up to 3 level, so not crazy), makes the code unreasonably noisy, in terms of writing custom encoder and decoders for repeated fields and sub-messages.

It would be really awesome if there was a switch to nanopb generator, that provides default encoder and decoders for everything, and uses standard malloc/free/realloc if needed, and puts the pointers (and length for arrays) in the generated structs directly.

This would remove SO MUCH boilerplate, and I think would be great for prototyping too.

If the design can be made so it also supports arena style allocation, that would be great, but optional.

PetteriAimonen, 2020-12-15:
Well.. I hope I'm not totally confused, but this sounds a lot like of two existing features:

- Static allocation: specify `max_count`, and the struct will have a statically allocated array of up to the maximum size.
- Dynamic allocation: specify `type = FT_POINTER` and enable `PB_ENABLE_MALLOC` and memory will be allocated using `pb_realloc()`, which can be overridden to use some custom allocator method.

The documentation is here: https://jpa.kapsi.fi/nanopb/docs/concepts.html#data-types
and perhaps these test cases may help also:

https://github.com/nanopb/nanopb/blob/master/tests/common/person.proto
https://github.com/nanopb/nanopb/blob/master/tests/mem_release/mem_release.proto

baryluk, 2020-12-15:
Ooo. I knew of `PB_ENABLE_MALLOC` (I already use it).

I didn't know of `type = FT_POINTER`.

Checking now.

baryluk, 2020-12-15:
@PetteriAimonen This totally works!

I even found a shortcut:  `nanopb_generator.py -s type:FT_POINTER my.proto`, to do it automatically to all repeated fields, nested stuff, and strings.

Nice.

Last question.

If I have a message that i want to encode, who takes responsibility of pointers?

I.e. can I do safely something like this:

```
   Message m = {0};
   int value = 42;
   m.value = &42;

   encode(...);

   pb_release(Message_fields, &m);

   ?
```

Probably a bad idea?

baryluk, 2020-12-15:
Hmm. Maybe if I use proto2, instead of proto3, it would be easier...

PetteriAimonen, 2020-12-15:
`pb_release()` assumes that pointers can be released with `pb_free()`. But one can use custom pointers safely when encoding, just don't call `pb_release()` or set the custom pointers to NULL before doing so.

And yeah, the distinction between `has_value` and implicit presence is a proto2 vs. proto3 thing. Though proto3 now also has an experimental feature to have `optional` specifier (issue #591).

I don't quite see the point of making all fields `FT_POINTER`. Instead, I would keep them at default (static allocation), and use FT_POINTER only for such repeated fields where you don't know a reasonable maximum size.

baryluk, 2020-12-15:
@PetteriAimonen I have a proto file with about 200 fields. I wish I would like to make it easy, and not annotate every field.

Maybe there is already a way to do so for string and repeated only?

PetteriAimonen, 2020-12-15:
No, currently there isn't such a way.

baryluk, 2020-12-15:
@PetteriAimonen Thank you and for the help.

closed 2020-12-15 by baryluk