nanopb/nanopb #624: Callback not in main message - submsg size changed error

KaDw, 2020-12-16:
I have a problem with following proto structure and nanopb-0.3.9.4:
```
syntax = "proto2";

message SocketSetup {
    optional SocketInfo info = 2;
}

message SocketInfo {
    repeated ProbeInfo      humidity = 1;
    repeated ProbeInfo      temperature = 2;
}

message ProbeInfo {
    enum PartNumber {
        R_55B1210 = 0;
    }
    optional PartNumber partNumber = 1;
    optional uint32 address      = 2 [default = 0]; 
    optional uint32 index        = 3 [default = 0];
}
```
My goal is to encode SocketSetup message. Since this callback is not directly in the main message It is firstly sized and then ProbeInfo is encoded. The sizing phase is correct and for two humidity and two temperature repeated messages firstly my callbacks are executed with empty stream structure, call stack:
[image]

After the sizing when nanopb tries to encode again it ends up in `PB_RETURN_ERROR(stream, "submsg size changed");` in `pb_encode_submessage()`

[image]

I have no problem when callback is directly in main message (eg. encode SocketInfo not SocketSetup).

Code to reproduce:
```
/*
this structure is shared across all repeated fields to
preserve context between callbacks
*/
typedef struct 
{
    uint16_t index;
    size_t encodedSize;
    bool sentAll;
    ProbeInfo *msgLastAddress;
    MeasurementsQueue_t *measQueueLastAddress;
}SocketInfoCbArg_t;

void SocketMgmtEncode(uint8_t *outBuf, size_t *outSize)
{
    static SocketInfoCbArg_t SocketInfoAnsCallbackArg = {
        .index = 0, .encodedSize = 0, .sentAll = true, .msgLastAddress = ProbeInfoHead, .measQueueLastAddress = NULL};
    pb_ostream_t ostream = pb_ostream_from_buffer(outBuf, *outSize);

    SocketInfoAnsCallbackArg.measQueueLastAddress = MeasurementsQueueGet();

    SocketSetupAns.info.humidity.funcs.encode = SocketInfo_humidity_callback;
    SocketSetupAns.info.humidity.arg = &SocketInfoAnsCallbackArg;
    SocketSetupAns.info.temperature.funcs.encode = SocketInfo_temperature_callback;
    SocketSetupAns.info.temperature.arg = &SocketInfoAnsCallbackArg;

    SocketSetupAns.has_info = true;

    if (!pb_encode(&ostream, SocketSetup_fields, &SocketSetupAns))
        {
            const char * error = PB_GET_ERROR(&ostream);
            printf("SocketSetup_fields error: %s\r\n", error);
            return false;
        }
}

static bool SocketInfoGenericCallback(pb_ostream_t *stream, const pb_field_t *field, SocketInfoCbArg_t *dest)
{
    if(dest->measQueueLastAddress->measurement->type != field->tag)
    {
        return true;
    }
    // first encode after sizing, init args again
    if(stream->callback != NULL && stream->bytes_written == 0)
    {
        SocketInfoResetCallbackArgs(dest);
    }

    while(1)
    {
        if(dest->measQueueLastAddress->measurement->type != field->tag)
        {
            break;
        }
        dst->has_partNumber = true;
        dst->has_address = true;
        dst->partNumber = src->measurement->id;
        dst->address = src->measurement->address;
        if(!pb_encode_tag_for_field(stream, field))
        {
            break;
        }

        if (!pb_encode_submessage(stream, ProbeInfo_fields, dest->msgLastAddress))
        {
            break;
        }
 
        dest->msgLastAddress++;
        dest->encodedSize = stream->bytes_written;

        if(dest->measQueueLastAddress->next == NULL || dest->msgLastAddress == NULL)
        {
            dest->sentAll = true;
            break;
        }
    
        dest->measQueueLastAddress = dest->measQueueLastAddress->next;
    }

    return true;
}

static bool SocketInfo_humidity_callback(pb_ostream_t *stream, const pb_field_t *field, void * const *arg)
{
    if (stream == NULL || field->tag != SocketInfo_humidity_tag)
    {
        return true;
    }

    return SocketInfoGenericCallback(stream, field, (SocketInfoCbArg_t*)(*arg));
}

static bool SocketInfo_temperature_callback(pb_ostream_t *stream, const pb_field_t *field, void * const *arg)
{
    
    if (stream == NULL || field->tag != SocketInfo_temperature_tag)
    {
        return true;
    }

    return SocketInfoGenericCallback(stream, field, (SocketInfoCbArg_t*)(*arg));
}
```

PetteriAimonen, 2020-12-16:
This part seems a bit questionable:

    // first encode after sizing, init args again
    if(stream->callback != NULL && stream->bytes_written == 0)
    {
        SocketInfoResetCallbackArgs(dest);
    }

Seems like during sizing run, only `SocketInfo_humidity_callback` will end up encoding data, because it runs first and the info is not reset when the stream is the sizing stream. However, during the actual encoding pass also `SocketInfo_temperature_callback` resets the state and encodes the same data again.
That's because `arg` points to the same structure for both - not sure if that is just in your example and not in reality.

I would suggest just restarting from beginning every time the `SocketInfoGenericCallback` runs. There is no need for an encoding callback to continue from where it left off, as it should just write out all data it has.

KaDw, 2020-12-16:
I see, thank you for fast reply.

You are right, arg points to the same structure. I have a limited space for protobuf output (in worst condition only 11 bytes). I thought passing around same structure which will have encodedSize variable that will keep last size of fully encoded message might save me some time instead of calling `pb_get_encoded_size` everytime (if its possible to call it inside callback)

Is there any other mechanism that will allow me to easly encode as much as I can fit to limited buffer?

PetteriAimonen, 2020-12-16:
I think I would approach that problem by encoding the items to a buffer until maximum number of bytes is reached. And only after that wrap the items inside the submessages and length prefixes - which can be done either manually following protobuf encoding spec, or alternatively by just having a callback copy them from a temporary buffer to the final buffer.

KaDw, 2020-12-16:
Sounds great, that simplifies a lot of my code! I will try to implement it and let you know

KaDw, 2020-12-16:
One more question, is there any compile time variable that determines number of fields in specific message?

PetteriAimonen, 2020-12-16:
Well, it's a bit of a hack, but in 0.3.x one can use `pb_arraysize(MyMessage_fields)-1` (-1 to subtract the list terminator). In 0.4.x there is `MyMessage_msg.field_count`.

KaDw, 2020-12-17:
I'm using LoRaWAN to send my messages. In LoRa when Adaptive Data Rate is enabled you can't control the payload size. You can get it only before sending a message. In my case messages are sent every 30 seconds

Unfortunately I'm unable to write a callback that will not have any common arguments, either global or passed via args struct - I must have last address of the ProbeInfo that I sent, I dont know where to. I also added index field based on repeated number and if there is more than 2 of the same kind, no index is sent for the later ProbeInfo - I can't predict how much space I will have and which ProbeInfo will have index or not, I encode it just before sending when I know how much space I have. It seems that I can't do that with pre-encoded approach. 

I don't have more time for this task so I edited the protobuf structure to make sure that there is no sizing involved in pb_encode.

closed 2021-03-22 by PetteriAimonen