nanopb/nanopb #625: Build static and dynamic libraries with PF_ENABLE_MALLOC and without

baryluk, 2020-12-17:
Hi,

first thanks for very good library. It basically just works, is well maintained and has comprehensive test suite.

I am developing a Linux desktop application / library, that uses `nanopb`, and it works really well.

However, there is an issue with some nanopb packages available in some distributions, in regards to `PF_ENABLE_MALLOC`.

For example in Debian: https://tracker.debian.org/pkg/nanopb  , `PF_ENABLE_MALLOC` is enabled ( see https://tracker.debian.org/media/packages/n/nanopb/rules-0.4.4-1 ) for static libraries.

But for example in Arch: https://aur.archlinux.org/packages/nanopb/ , `PF_ENABLE_MALLOC` is NOT enabled (see https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=nanopb ) for static libraries. So if the app is enabling `PF_ENABLE_MALLOC` and calls `pb_release` (as it should most of the time), linking with `-lprotobuf-nanopb` or `/usr/lib/libprotobuf-nanopb.a` (or similar), will fail during linking with `pb_release` undefined / not found. That is expected, but not good user experience.

This causes issues to any portable project that want to link to distro provided libraries. (Using distro provided libraries and headers is just simple for the user, than pulling dependencies or keeping local copy of nanopb).

My question is, could `nanopb` just build both versions autoamtically in `cmake`? `libprotobuf-nanopb.a` and `libprotobuf-nanopb-malloc.a`. That would help a lot. I know this can be done on per-distro basis to do so, but that is the problem, because some distros will then provide incompatible versions for sure.

(Same for shared library versions).

Thank you for your consideration.

PetteriAimonen, 2020-12-17:
Well, to be honest, I have no idea what to do about this. I'm not really targeting nanopb for Linux, and the design is especially unsuitable for dynamic libraries - ABI breaking changes occur regularly.

To reduce my own workload, I would prefer just to say "don't use nanopb as a dynamic library". But people will anyway. And as you say, the problem of compilation options exists for static libraries also.

On the other hand, there is no real reason not to enable `PB_ENABLE_MALLOC` by default in all packages built for Linux. It's disabled by default because the main target for nanopb is microcontrollers with mere kilobytes of RAM.

baryluk, 2020-12-17:
@PetteriAimonen I do understand why it is not enabled by default. I don't have issues with that.

Also, I don't need dynamic library really.

Do you think the default for `PB_ENABLE_MALLOC` should be change somehow in nanopb in this repo, or is this up to the Linux distros?

PetteriAimonen, 2020-12-17:
I think it would be up to the distros, as I do not want to enable it by default for everyone.

baryluk, 2020-12-17:
Then that is a bit of problem, because some distro maintainers would simply not set it, and various distros will ship it in different state.

That is why I proposed to build both and have them have different suffix.

PetteriAimonen, 2020-12-17:
Yeah. I myself would even be fine with having `PB_ENABLE_MALLOC` on by default with CMake builds, as that is mostly used on Linux AFAIK. Building multiple libraries with various names sounds cumbersome.

@dflogeras, @der-imp, @mrostecki, @podsvirov Any thoughts?

dflogeras, 2020-12-22:
IMHO Classic case of "default tug-of-war" where no matter what, someone will want the opposite.  I personally use the CMake build to cross compile and target my ARM uC systems.  

However, if for the greater good, making the default set PB_ENABLE_MALLOC to true solves more problems (wider audience), I'd say it's a reasonable compromise.  Someone like me is far more likely to tailor the build than a distro maintainer who just builds the package.

All that said, you will still likely run into distros shipping with it built in ways conflicting with your project/use case.  For that, I'd say is a distro problem and not a nanopb problem.  I know my distro has many such packages that have to be built in multiple permutations (for example providing both 64 and 32 bit libs), or have a xyz-compat package to satisfy some condition for legacy binary programs (such as ncurses/openssl).

Et7f3, 2022-08-12:
(I don't need PB_ENABLE_MALLOC by default) Could a check be added that test availability and if the platform support it then it enable it. So if builded with a cross compiler check on malloc will fail and then it will be disabled unless explicitly defined (in case of custom allocator).