nanopb/nanopb #626: [QUESTION] ESP8266 with NanoPB and gRPC

mateusscheper, 2020-12-19:
Hi! 

I'm learning how to make an API with golang (server side) and I'd like to make my ESP8266 as a client. 
The problem is, I can't quite figure out how to call the server gRPC functions. So far, I can only call my server URL, like a normal API.
Question: is it possible to call functions instead of passing my requests via URL?

This is my proto file:
```
syntax = "proto3";

message TemperaturaEvento {
  uint64 data_hora = 1;
  float temperatura = 2;
}

message PressaoEvento {
  uint64 data_hora = 1;
  float pressao = 2;
}

message TemperaturaResponse {
  bool ok = 1;
}

service AmbienteService {
  rpc RegistraTemperatura(TemperaturaEvento) returns (TemperaturaResponse) {}
}
```

And this is how I'm calling my server:
```
void registraTemperatura() {
    if (!client.connect(addr, port)) {
       Serial.println("Falha ao conectar-se no servidor " + String(addr) + ", porta " + port);
    }
    TemperaturaEvento medicao = TemperaturaEvento_init_zero;
    timeClient.update();
    Serial.println(timeClient.getEpochTime());
    medicao.data_hora = timeClient.getEpochTime();
    medicao.temperatura = float(150);

    uint8_t buffer[128];
    pb_ostream_t stream = pb_ostream_from_buffer(buffer, sizeof(buffer));

    if (!pb_encode(&stream, TemperaturaEvento_fields, &medicao)) {
        Serial.println("Falha ao enviar temperatura para a API.");
        return;
    }
    client.write(buffer, stream.bytes_written);
    delay(1000);
}
```
(Don't use String, I know)

I know this is not a NanoPB question exactly, but I can't find anything on Google. 

Ps: I'm still a noob.

PetteriAimonen, 2020-12-20:
I'm not very familiar with gRPC either, but there is something here that d21d3q has been working on: https://github.com/d21d3q/nanogrpc

closed 2021-03-22 by PetteriAimonen

gentry-tran, 2021-11-30:
@mateusscheper did you ever figure this out? I am running into this same issue.