nanopb/nanopb #627: Nanopb cannot communicate with google protobuf

mejtejt, 2020-12-23:
I have two sides of application. One side is using google protobuf c++ (version 3.11.2), and the other side nanopb (version 0.4.5).

This is the proto file on C++ side: 
[image]

Part of the code which serializes this is as follows:
[image]

Payload is then sent over UART (part in the brackets [0x8 0x1]):
[image]

On the nanopb side, this message is then received. Following are .proto file and serial output:
[image]
[image]

`Payload data` shows what has been sent from C++ side and received on the nanopb side. The problem is, if I try to decode this using `pb_decode` function, it fails with "wrong wire type" error message. 

The next step in trying to identify the problem was the following: I will encode MessageRequest on the nanopb side, and print out the bytes to see what is makes out of it. 
[image]
[image]

As you can see, `Payload data =` is the part received from C++ protobuf. And `Request buffer =` is what comes out after calling pb_encode function. Now decoding, on line 65, works. 

The question is: what is `0xa` and `0x2` part of the request buffer? And why does google protobuf not generate this? This is the part that is making `pb_decode` function to fail. How do I make them compatible (either by making google protobuf generating this (I don't know what it is, so I will call it) header, or to instruct nanopb to not use this during encoding/decoding)?

PetteriAimonen, 2020-12-23:
First of all, are you using different `.proto` files for encoding and decoding? Why? They should match on each side.

It also appears you are encoding `MetricPair` but trying to decode `MessageRequest`. Because these two message types have different and incompatible structures, decoding will fail. In this case `0x0A 0x02` is the submessage prefix.

closed 2020-12-23 by PetteriAimonen

mejtejt, 2020-12-23:
Ahhh, I am so sorry for taking your time on this. You are absolutely right, they should be the same. I am not the only one working on this, that is why they are incompatible (at the moment). But, as you said, they should be identical. 

I have been staring at this so long, that I totally missed what I am encoding is different from what I am decoding. Thanks a lot for a quick response.

PetteriAimonen, 2020-12-23:
No problem, good to hear you got it working!