nanopb/nanopb #628: How to use Proto3 Optional with nanopb_generator?

Awbmilne, 2021-01-13:
I am looking to compile proto3 files with `nanopb_generator`, but have run into this error message:

```
myProto3.proto: This file contains proto3 optional fields, but --experimental_allow_proto3_optional was not set.
```

Based on the architecture of this tool, this seems to me like the `--experimental_allow_proto3_optional` flag is not being passed along as it should be.

**Is there a way to either:**
- **Declare this flag when calling `nanopb_generator`**
- **Call `protoc` with `nanopb` as an extension so I can add the flag myself**

So far, I have tried:
- Adding `--experimental_allow_proto3_optional` to the `nanopb_generator` call
- Adding `-s PROTOCFLAGS:--experimental_allow_proto3_optional` to the `nanopb_generator` call
- Adding `export PROTOCFLAGS "--experimental_allow_proto3_optional"` as an environment variable before call to `nanopb_generator`

---

Im sure this is just me not understanding python argument syntax, But im pretty hung up here.

I have been refering to `tests/proto3_optional/SConscript` to try understand how to implement this:\
https://github.com/nanopb/nanopb/blob/a94b77af2b2113bc64acb6236290d7b64461b2eb/tests/proto3_optional/SConscript#L1-L23

Thanks for the great tools!

PetteriAimonen, 2021-01-13:
Oh, I haven't thought about the need to pass options to `protoc` when calling it through `nanopb_generator`. I should probably add something for that.

In the meanwhile, these ways should work:

- Two steps: `protoc --experimental_allow_proto3_optional -omyfile.pb myfile.proto` `nanopb_generator myfile.pb`
- Calling through protoc: `protoc --nanopb_out=outdir myfile.proto`

For the latter you'll need `protoc-gen-nanopb` in your path, or to pass the path to it with `--plugin` to `protoc`.

There is a long history behind this. Initially `protoc` didn't support plugins, so the two-step method was the way to go. Then plugin support was added and I started recommending it, but it made passing options to `nanopb_generator` difficult and complicated handling of relative paths. Then in 2019 I reconsidered and made it possible for `nanopb_generator` to call `protoc`, which made it easier to pass options to `nanopb_generator`. And now we've come the full circle and have to pass `protoc` options through `nanopb_generator` :)

PetteriAimonen, 2022-11-25:
Added `--protoc-opt` argument to `nanopb_generator.py` now.
Recent versions of `protoc` don't need `--experimental_allow_proto3_optional` anymore.

Awbmilne, 2022-12-06:
I do not use nanopb on a regular basis anymore. As such I do not have a quick way to test that this is a resolved issue. I will leave closing this issue at your discretion. 👍

closed 2022-12-11 by PetteriAimonen