nanopb/nanopb #629: Python 2.7 vs Python 3? [Gitlab CI]

Awbmilne, 2021-01-14:
I have recently been working on configuring GitLab CI to build Protocol Buffers files for use as a library.

I seem to have to have it working now, but I struggled a lot to get here.

Along with installing protoc, I used the nanopb python package. \
It seemed to install fine on both Python 2.7 and Python 3.8, But failed to run as a plugin for protoc.

Here is the config file\
`.gitlab-ci.yml`:
```yml

stages:
  - build

image: python:2.7 # 'python:3.8', 'python:3.9', and 'python:3.9-buster' dont work

compile_protocol_buffers:
  stage: build
  before_script:
    - PROTOC_ZIP=protoc-3.14.0-linux-x86_64.zip
    - curl -OL https://github.com/protocolbuffers/protobuf/releases/download/v3.14.0/$PROTOC_ZIP
    - unzip -o $PROTOC_ZIP -d /usr/local bin/protoc
    - unzip -o $PROTOC_ZIP -d /usr/local 'include/*'
    - rm -f $PROTOC_ZIP
    - pip install nanopb
    - mkdir ~/protobuf     || true
    - mkdir ./gen          || true
    - mkdir ./gen/cpp      || true
    - mkdir ./gen/objc     || true
    - mkdir ./gen/python   || true
    - mkdir ./gen/embedded || true
  script:
    - protoc
      --proto_path=protos
      --cpp_out=gen/cpp
      --objc_out=gen/objc
      --python_out=gen/python
      --nanopb_out=gen/embedded
      --plugin=nanopb_generator
      --experimental_allow_proto3_optional
      protos/main.proto

```

When running with either `python:3.8` or `python:3.9`, I get this error:

```bash
$ protoc --proto_path=protos --cpp_out=gen/cpp --objc_out=gen/objc --python_out=gen/python --nanopb_out=gen/embedded --plugin=nanopb_generator --experimental_allow_proto3_optional protos/main.proto
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/nanopb/generator/nanopb_generator.py", line 51, in <module>
    from .proto import nanopb_pb2
ImportError: cannot import name 'nanopb_pb2' from 'nanopb.generator.proto' (/usr/local/lib/python3.8/site-packages/nanopb/generator/proto/__init__.py)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/bin/protoc-gen-nanopb", line 5, in <module>
    from nanopb.generator.nanopb_generator import main_plugin
  File "/usr/local/lib/python3.8/site-packages/nanopb/generator/nanopb_generator.py", line 72, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
ModuleNotFoundError: No module named 'proto'
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

This doesnt seem to be an issue with my Windows machine as I only have python 3.8+ installed and it works fine.\
But something about the Python docker images doesnt agree with `nanopb_generator`.

Not a pressing issue for myself, But a weird bug that I have had to work around.\
Would be great if I could use python 3.8-3.9 as 2.7 has passed end-of-life, But not a big deal.

Thanks for the great work!

PetteriAimonen, 2021-01-15:
I can reproduce the bug, but only when installing system-wide. Looks like `nanopb_pb2.py` is not getting added during installation for some reason.

As a workaround, one can run `nanopb_generator` once as root, and it will generate the file. After that it will work for normal users also.

PetteriAimonen, 2021-03-22:
Fix released in 0.4.5.

closed 2021-03-22 by PetteriAimonen

ghpqans, 2023-06-20:
I now ran into this issue using version 0.4.7:
```
python3 -V
Python 3.10.11

pip3 list -l
Package      Version
------------ -------
cffi         1.15.1
google       3.0.0
grpcio       1.54.2
grpcio-tools 1.54.2
nanopb       0.4.7
pip          23.0.1
protobuf     4.23.3
pycparser    2.21
setuptools   65.5.0
```

I generated a python file with the nanopb message definitions using:
```
.venv/lib/python3.10/site-packages/nanopb/generator/protoc -I=in_path -I=.venv/lib/python3.10/site-packages/nanopb/generator/proto --python_out=. in_path/nanopb_defs.proto 
```

The generated file contains the following imports:
```
# Generated by the protocol buffer compiler.  DO NOT EDIT!
# source: bb_serialize_dsl.proto

import sys
_b=sys.version_info[0]<3 and (lambda x:x) or (lambda x:x.encode('latin1'))
from google.protobuf.internal import enum_type_wrapper
from google.protobuf import descriptor as _descriptor
from google.protobuf import message as _message
from google.protobuf import reflection as _reflection
from google.protobuf import symbol_database as _symbol_database
# @@protoc_insertion_point(imports)

_sym_db = _symbol_database.Default()

import nanopb_pb2 as nanopb__pb2
```
This line `import nanopb_pb2 as nanopb__pb2` 
results in the error:
```
/home/user/.../nanopb_defs_pb2.py 
Traceback (most recent call last):
  File "/home/user/.../nanopb_defs_pb2.py", line 16, in <module>
    import nanopb_pb2 as nanopb__pb2
ModuleNotFoundError: No module named 'nanopb_pb2'
```

So I vote for reopening this Issue.

PetteriAimonen, 2023-06-20:
@ghpqans To me your comments sound like a different issue than what this one was about. So let's continue in #877.

ghpqans, 2023-06-20:
```
.venv/lib/python3.10/site-packages/nanopb/generator/protoc --version
libprotoc 3.21.12
```