nanopb/nanopb #635: init_default not populating default values when using oneof

zorce, 2021-02-03:
Hi,

Have some problem when initialize with init_default for a message with oneof signals. Then the default values are not expanded for the payload message.
`syntax = "proto2";

message Configure {
    optional uint32 parameter1 = 1 [default = 24];
    optional uint32 parameter2 = 2 [default = 1];
}

message EnableStuff {
    optional uint32 enable = 1 [default=1];
}

/* ---- Request signals ---- */
message Request {
    oneof payload {
        Configure configure = 1;
        EnableStuff enable_stuff = 2;
    }
}
`

When decoding I do the following
`// Allocate the request message on the stack
Request req = **Request_init_default**;

// Create a stream to read from our buffer
pb_istream_t stream = pb_istream_from_buffer(buffer, buffer_size);

// Decode our message
status = pb_decode(&stream, Request_fields, &req);

// check if decoding was ok
if (!status) {
	return 1;
}`

How should I unpack to also get the default values without creating another object for the payload signal and campare them?

PetteriAimonen, 2021-02-03:
Hmm, the `init_default` cannot initialize default values inside `oneof` because the storage is shared. However `pb_decode()` should do that.

This appears to be a regression caused by commit edf6dcbffee in versions 0.4.4 and possibly 0.3.9.7 also.

PetteriAimonen, 2021-02-03:
Should be fixed in git now. The 0.3 branch was not affected, but added the test case there also.

zorce, 2021-02-24:
Perfect! Thanks alot! Will probably wait for a generated release on this fix and set all the variables until then.

PetteriAimonen, 2021-03-22:
Fix released in 0.4.5.

closed 2021-03-22 by PetteriAimonen