nanopb/nanopb #636: Size of encoded array is insane

AABelkhiria, 2021-02-03:
Hello

We are working on a project where we need to send data from node to node and we decided to migrate to nanopb.
All messages have this layout
```
message MSG_MESSAGE {
  required MSG_HEADER header = 1;
  repeated uint32 payload = 2;
}
```
where
```
MSG_MESSAGE.payload type:FT_STATIC, max_count:1024, int_size:IS_8
```
Everything works fine but we have noticed that the size of the packet is now nearly 3 times and it is also mentioned inside msg_message.pb.h
```
/* Maximum encoded size of messages (where known) */
#define MSG_MESSAGE_size                         3086
```
where
```
/* Maximum encoded size of messages (where known) */
#define MSG_HEADER_size                          12
```

How can we encode the data (serialize it) without getting the size tripled ?

PetteriAimonen, 2021-02-03:
nanopb will encode the array in the packed format, which will take less space (1 byte per value up to 127, 2 bytes per value from there). However because the protobuf format allows unpacked representation (2 bytes per value up to 127, 3 bytes per value from there), the size calculation in header reserves the worst case size.

Maybe you could use the `bytes` type instead, if your payload consists of `uint8_t` values. That will be more space efficient (constant 1 byte per value).

AABelkhiria, 2021-02-04:
That's perfect
Moving to
```
message MSG_MESSAGE {
  required MSG_HEADER header = 1;
  required bytes payload = 2;
}
```
reduced the size to
```
/* Maximum encoded size of messages (where known) */
#define MSG_MESSAGE_size                         1017
```

Thanks this was really helpful 🙏

closed 2021-02-04 by AABelkhiria