nanopb/nanopb #637: Generator produces incorrect output when a `oneof` is used

rileylyman, 2021-02-10:
This issue has been successfully recreated using the `0.4.3` and `0.4.4` tags as well as the latest `master` commit.

I ran into this strange issue today and put together a minimal recreation project here: https://github.com/rileylyman/nanopb_union_issue. If you run Linux and have `gcc` installed, you can run this out of the box with the `minimal_example.sh` script and see the error.

In the repository I linked are the following files:

Main.proto:
```
syntax = "proto3";

import "Other.proto";

message FirstOneof {}

message Bar {
  oneof content {
    FirstOneof first = 1;
    SecondOneof second = 2;
  }
}

message Foo {
  Bar bar = 1;
}

```

Other.proto:
```
syntax = "proto3";

message SecondOneof {}

```

When I run `nanopb_generator.py` on `Main.proto`, I get the following output:
```
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.5-dev */                                                                                                           
                                                                                                                                              
#ifndef PB_MAIN_PB_H_INCLUDED                                          
#define PB_MAIN_PB_H_INCLUDED
#include <pb.h>                             
#include "Other.pb.h"                         

#if PB_PROTO_HEADER_VERSION != 40           
#error Regenerate this file with the current version of nanopb generator.
#endif                                                                                                                                        
                                                                                                                                              
/* Struct definitions */                            
typedef struct _FirstOneof {                                                                                                                  
    char dummy_field;                                                                                                                         
} FirstOneof;                                                          
                                                                       
typedef struct _Bar {                             
    pb_size_t which_content;                                                                                                                  
    union {                                       
        FirstOneof first;                                              
        SecondOneof second;                         
    } content;                      
} Bar;                                                                 
                                                                       
typedef struct _Foo {          
    bool has_bar;                                                                                                                             
    Bar bar;                                                                                                                                  
} Foo;                                                                                                                                        
                                                                                                                                              
                                                                                                                                              
#ifdef __cplusplus                                                                                                                            
extern "C" {                                
#endif                                        
                                                                       
/* Initializer values for message structs */
#define FirstOneof_init_default                  {0}    
#define Bar_init_default                         {0, {FirstOneof_init_default}}
#define Foo_init_default                         {false, Bar_init_default}
#define FirstOneof_init_zero                     {0}                      
#define Bar_init_zero                            {0, {FirstOneof_init_zero}}
#define Foo_init_zero                            {false, Bar_init_zero}

/* Field tags (for use in manual encoding/decoding) */
#define Bar_first_tag                            1
#define Bar_second_tag                           2
#define Foo_bar_tag                              1

/* Struct field encoding specification for nanopb */
#define FirstOneof_FIELDLIST(X, a) \

#define FirstOneof_CALLBACK NULL
#define FirstOneof_DEFAULT NULL

#define Bar_FIELDLIST(X, a) \
X(a, STATIC,   ONEOF,    MESSAGE,  (content,first,content.first),   1) \
X(a, STATIC,   ONEOF,    MESSAGE,  (content,second,content.second),   2)
#define Bar_CALLBACK NULL
#define Bar_DEFAULT NULL
#define Bar_content_first_MSGTYPE FirstOneof
#define Bar_content_second_MSGTYPE SecondOneof

#define Foo_FIELDLIST(X, a) \
X(a, STATIC,   OPTIONAL, MESSAGE,  bar,               1)
#define Foo_CALLBACK NULL
#define Foo_DEFAULT NULL
#define Foo_bar_MSGTYPE Bar

extern const pb_msgdesc_t FirstOneof_msg;
extern const pb_msgdesc_t Bar_msg;
extern const pb_msgdesc_t Foo_msg;

/* Defines for backwards compatibility with code written before nanopb-0.4.0 */
#define FirstOneof_fields &FirstOneof_msg
#define Bar_fields &Bar_msg
#define Foo_fields &Foo_msg

/* Maximum encoded size of messages (where known) */
#define FirstOneof_size                          0
#if defined(SecondOneof_size)
union Bar_content_size_union {char f2[(6 + SecondOneof_size)]; char f0[2];};
#define Bar_size                                 (0 + sizeof(union Bar_content_size_union))
#endif
#if defined(SecondOneof_size)
union Bar_content_size_union {char f2[(6 + SecondOneof_size)]; char f0[2];};
#define Foo_size                                 (6 + sizeof(union Bar_content_size_union))
#endif

#ifdef __cplusplus
} /* extern "C" */
#endif

#endif
```
As you can see, at the end of this generated output, `Bar_content_size_union` is defined twice, causing the `gcc` compilation to fail. 

```
In file included from Main.pb.c:4:
Main.pb.h:85:7: error: redefinition of ‘union Bar_content_size_union’
   85 | union Bar_content_size_union {char f2[(6 + SecondOneof_size)]; char f0[2];};
      |       ^~~~~~~~~~~~~~~~~~~~~~
Main.pb.h:81:7: note: originally defined here
   81 | union Bar_content_size_union {char f2[(6 + SecondOneof_size)]; char f0[2];};
      |       ^~~~~~~~~~~~~~~~~~~~~~
```

This only happens if `SecondOneof` message is defined in another file. If you move the `message SecondOneof {}` into `Main.proto`, then the generator works fine. 

This is a very strange issue, I hope that what I have provided is useful!

PetteriAimonen, 2021-02-11:
Hmm yeah, needs some logic to avoid emitting the same declaration multiple times. Yet another regression from #415.

As a workaround, it may work better if you pass all files to the generator at the same time (i.e. list all .proto files in a single command).

tymmej, 2021-03-20:
Hi,

I fix it for my own use.
Here is commit with explanation what I did: https://github.com/tymmej/nanopb/commit/9a907d70e863cd3f10c72c586a7aa90f97fa158a

I've created pull request if you feel it's right approach: https://github.com/nanopb/nanopb/pull/648

PetteriAimonen, 2021-03-22:
@tymmej Thanks, your fix is now merged.

PetteriAimonen, 2021-03-22:
Fix released in 0.4.5.

closed 2021-03-22 by PetteriAimonen