nanopb/nanopb #638: Using the google wrappers.proto and descriptor.proto with nanopb.proto

Yatekii, 2021-02-11:
Hi!

Thanks for the hard work on this lib :) It has been doing good work for me until recently when I tried generating the API surface with `google/protobuf/descriptor.proto` and `google/protobuf/wrappers.proto`. I am using `nanopb.proto` which uses `google/protobuf/descriptor.proto` itself and I use `google/protobuf/wrappers.proto` at some other place.
Having the google includes itself I get:

```
++ python3 scripts/generator/nanopb_generator.py -e .npb '--library-include-format=#include <z%s>' -I=./src -I=./src/addressing -I=./src/attributes --output-dir=./generated_nanopb ./src/addressing/automation_addressing.proto ./src/addressing/blinds_addressing.proto ./src/addressing/channel_addressing.proto ./src/addressing/device_addressing.proto ./src/addressing/front_surface_addressing.proto ./src/addressing/group_addressing.proto ./src/addressing/light_addressing.proto ./src/addressing/network_addressing.proto ./src/addressing/peripheral_addressing.proto ./src/addressing/scene_addressing.proto ./src/attributes/automation_attributes.proto ./src/attributes/blinds_attributes.proto ./src/attributes/device_attributes.proto ./src/attributes/front_surface_basic_attributes.proto ./src/attributes/group_attributes.proto ./src/attributes/light_attributes.proto ./src/attributes/network_attributes.proto ./src/attributes/peripheral_attributes.proto ./src/attributes/scene_attributes.proto ./src/addressing.proto ./src/address_resolution.proto ./src/attributes.proto ./src/authentication.proto ./src/automation.proto ./src/ble.proto ./src/blinds.proto ./src/channel.proto ./src/client.proto ./src/commands.proto ./src/config.proto ./src/device.proto ./src/dfu.proto ./src/discovery.proto ./src/event.proto ./src/factory_testing.proto ./src/front_surface_basic.proto ./src/front_surface.proto ./src/functions.proto ./src/general.proto ./src/group.proto ./src/heartbeat.proto ./src/homekit.proto ./src/identify.proto ./src/iot.proto ./src/light.proto ./src/message.proto ./src/nanopb.proto ./src/network.proto ./src/peripheral.proto ./src/product.proto ./src/scene.proto ./src/script.proto ./src/status.proto ./src/subscription.proto ./src/thread.proto ./src/timer.proto ./src/zigbee.proto ./src/zigfred_plus.proto ./src/zigfred_uno.proto
nanopb.proto:145:12: "google.protobuf.FieldDescriptorProto.Type" is not defined.
nanopb.proto:164:8: "google.protobuf.FileOptions" is not defined.
nanopb.proto:168:8: "google.protobuf.MessageOptions" is not defined.
nanopb.proto:172:8: "google.protobuf.EnumOptions" is not defined.
nanopb.proto:176:8: "google.protobuf.FieldOptions" is not defined.
zigbee.proto:4:1: Import "nanopb.proto" was not found or had errors.
thread.proto:4:1: Import "nanopb.proto" was not found or had errors.
iot.proto:4:1: Import "zigbee.proto" was not found or had errors.
iot.proto:5:1: Import "thread.proto" was not found or had errors.
iot.proto:9:3: "zigbee.ZigbeeAddress" is not defined.
iot.proto:10:3: "thread.ThreadAddress" is not defined.
addressing/device_addressing.proto:4:1: Import "iot.proto" was not found or had errors.
addressing/device_addressing.proto:8:3: "iot.IoTAddress" is not defined.
```

I tried vendoring the descriptor and wrappers and putting them directly into my source directory. That works well, but a) it is ugly, because I manually manipulate library files and b) when you have the google python protobuf installed it becomes a namespace clash. So I tried using another namespace. That does not work because then protoc itself will not accept the Option extensions of nanopb :/

Do you have an idea how I do this properly? I think I tried pretty much everything. I am at the point now where I do two sets of definitions (one for python and one for nanopb) which is basically the point where I throw everythign into the garbage because at this point protobuf is nowhere near sane anymore :/

I know that the python part does not tangent this library but nanopb apparently not being able to build its own nanopb.proto surely is :)

Thanks a lot :)

Yatekii, 2021-02-11:
I just unvendored everything properly, which gives me:

```
Traceback (most recent call last):
  File "scripts/generator/nanopb_generator.py", line 1449, in default_value
    enumtype = dependencies[str(enumname)]
KeyError: 'google_protobuf_FieldDescriptorProto_Type'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "scripts/generator/nanopb_generator.py", line 2518, in <module>
    main_cli()
  File "scripts/generator/nanopb_generator.py", line 2394, in main_cli
    results = process_file(fdesc.name, fdesc, options, other_files)
  File "scripts/generator/nanopb_generator.py", line 2299, in process_file
    headerdata = "".join(f.generate_header(includes, headerbasename, options))
  File "scripts/generator/nanopb_generator.py", line 1811, in generate_header
    yield msg.fields_declaration(self.dependencies) + "\n"
  File "scripts/generator/nanopb_generator.py", line 1317, in fields_declaration
    defval = self.default_value(dependencies)
  File "scripts/generator/nanopb_generator.py", line 1451, in default_value
    raise Exception(
Exception: Could not find enum type google_protobuf_FieldDescriptorProto_Type while generating default values for NanoPBOptions.
Try passing all source files to generator at once, or use -I option.
```

Which is odd, because it can find the .proto file, it just does not generate anything from it. So the question is now how I tell it to generate the file too.

Passing `-I="./scripts/generator/proto/google/protobuf" \` (for wrappers.proto and descriptor.proto) results in:

```
Traceback (most recent call last):
  File "scripts/generator/nanopb_generator.py", line 1449, in default_value
    enumtype = dependencies[str(enumname)]
KeyError: 'google_protobuf_FieldDescriptorProto_Type'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "scripts/generator/nanopb_generator.py", line 2518, in <module>
    main_cli()
  File "scripts/generator/nanopb_generator.py", line 2394, in main_cli
    results = process_file(fdesc.name, fdesc, options, other_files)
  File "scripts/generator/nanopb_generator.py", line 2299, in process_file
    headerdata = "".join(f.generate_header(includes, headerbasename, options))
  File "scripts/generator/nanopb_generator.py", line 1811, in generate_header
    yield msg.fields_declaration(self.dependencies) + "\n"
  File "scripts/generator/nanopb_generator.py", line 1317, in fields_declaration
    defval = self.default_value(dependencies)
  File "scripts/generator/nanopb_generator.py", line 1451, in default_value
    raise Exception(
Exception: Could not find enum type google_protobuf_FieldDescriptorProto_Type while generating default values for NanoPBOptions.
Try passing all source files to generator at once, or use -I option.
```

PetteriAimonen, 2021-02-11:
Hmm, do you actually *want* to generate C definitions for the `descriptor.proto` and `nanopb.proto`? Are you going to access them directly from C code? By default they should be in the exclude list, i.e. the nanopb generator just won't #include the .h file for them.

As for include path, I think you should pass `-I` path to the folder that contains the `google` folder. The "official name" for `descriptor.proto` is `google/protobuf/descriptor.proto` and that should be used when including it in `.proto` files.

Yatekii, 2021-02-11:
I do not need to access them from within C. So I guess I do not need them really.

But, where does it get all the type references from then? I mean they have to be somewhere. That's what I do not understand :/
I mean, the generator tells me that there is a type missing. It's not like I could do much about that, right?

```
set -o xtrace

OUTPUT_FOLDER="./build/generated_nanopb"
rm -r $OUTPUT_FOLDER
mkdir -p $OUTPUT_FOLDER

protoc --version

python3 ./scripts/generator/nanopb_generator.py \
    -e .npb \
    --library-include-format="$incl_str" \
    -I=./src \
    -I="./src/addressing" \
    -I="./src/attributes" \
    -I="./scripts/generator/proto/google/protobuf" \
    --output-dir=$OUTPUT_FOLDER \
    ./scripts/generator/proto/google/protobuf/*.proto ./src/**/*.proto ./src/*.proto
```

This is my current command. I do not know how I could leave away anything else :/

Yatekii, 2021-02-11:
The problem is basically that

```proto
syntax = "proto3";
package channel;

import "general.proto";
import "addressing/peripheral_addressing.proto";
import "google/protobuf/wrappers.proto";

message ChannelConfig {
  general.Time timestamp = 1;
  bool enabled = 2;
  oneof mode {
    peripheral_addressing.PeripheralLink peripheral_link = 3; // Will generate a peripheral if set with same PeripheralID as the ChannelID
    bool bypass = 4; // Will turn on this channel always (e.g. for Zigbee Smart Light Bulbs on the Wire, which have to be always on and have their own radio interface)
  }
}

message ChannelStatus {
  uint32 index = 1;
  google.protobuf.BoolValue overcurrent = 2;
  google.protobuf.BoolValue drawing_current = 3;
  google.protobuf.UInt32Value measured_level = 5;
  google.protobuf.UInt32Value measured_current = 6;
}
```

results in a `channel.pb.h` with `#include "google/protobuf/wrappers.pb.h"` in it.

PetteriAimonen, 2021-02-11:
Yes I agree that you need `wrappers.pb.h` for the types. I mean that you probably don't need `nanopb.pb.h` or `descriptor.pb.h`.

> -I="./scripts/generator/proto/google/protobuf"

try changing that to `-I="./scripts/generator/proto"` if you have wrappers.proto available there - otherwise pass a path to where you have it.

Yatekii, 2021-02-11:
I have managed to fix it by compiling the wrappers and generators separately with

```
python3 ./scripts/generator/nanopb_generator.py \
    -e .pb \
    -I="./scripts/generator/proto/google/protobuf" \
    --output-dir="$OUTPUT_FOLDER/google/protobuf" \
    ./scripts/generator/proto/google/protobuf/*.proto
```

And then in my C build I include those files manually. Seems to work. If I encounter anymore issues, I will report back. I guess until then we can consider this closed :) But, maybe we should plant a hint somewhere how one makes this work. This cost me quite a lot of time now. But most of it was because of C being garbage anyways, so this might not matter too much :)

Thanks a ton for your help! Much appreciated :)

closed 2021-02-11 by Yatekii