nanopb/nanopb #639: Help in configuring Atmel Studio 7

Zen89, 2021-02-17:
Hello,
I have a problem configuring the Atmel Studio 7 with nanopb. I mean the toolchain and makefile configuration. I don't know where to enter and change what to make everything work properly. Please give me some advice, I would be very grateful for instructions with screenshots. At the moment working with python and json files is black magic to me.

Thank you very much and have a nice day.

PetteriAimonen, 2021-02-17:
Sorry, I have no experience with that IDE and I don't have resources to provide free support for it.

closed 2021-02-17 by PetteriAimonen

Zen89, 2021-02-17:
Okay, thank you for your answer. Could you direct me where to look for knowledge to come to the correct configuration myself?

apcountryman, 2021-02-17:
The documentation for Atmel Studio 7 (now Microchip Studio 7) can be found [here](https://onlinedocs.microchip.com/pr/GUID-ECD8A826-B1DA-44FC-BE0B-5A53418A47BD-en-US-6/index.html?GUID-00257F02-E33C-40C3-B324-83DBCC05EC30). I don't use Microchip Studio 7 so I can't help you with this, but [avrfreaks](https://www.avrfreaks.net/) is a great forum for questions relating to AVRs and the associated tools.

Zen89, 2021-02-17:
Thank you very much!