nanopb/nanopb #640: Generating size defines for max_count arrays

zorce, 2021-02-24:
Hi,
More of a feature request but it would be nice to have defines for the fixed array sizes instead of just generating the size on the message struct. That would make it alot easier to create buffers based on the generated protobuf.

From this:
```
typedef struct _MyMessage {
    bool just_an_example;
    pb_size_t my_array_count;
    int32_t my_array[9];
} MyMessage;
```
To this:
```
typedef struct _MyMessage {
    bool just_an_example;
    pb_size_t my_array_count;
    int32_t my_array[my_array_max_count];
} MyMessage;
#define my_array_max_count 9
```

PetteriAimonen, 2021-02-24:
In most cases you can use `pb_arraysize` macro, like this:

    int data[pb_arraysize(MyMessage, my_array)];

zorce, 2021-02-24:
Ah perfect! Then no need, didn't notice that macro. Thanks!

closed 2021-02-24 by zorce