nanopb/nanopb #641: interoperation with swift protobuf issue: oneof messages is not decoded properly

truebucha, 2021-02-24:
Hi! I have an issue with sending data encoded with nanopd to mobile that decoded it using [swift protobuf](https://github.com/apple/swift-protobuf).

So general question: is nanopb supposed to work well when decoded by one of other protobuf decoders?

The mobile with swift protobuf failed to decode oneof section. It succeeds with two uint32 fields that precede the oneof field. But oneof field failed to be decoded and it's data was moved to unknownField. I try to create the message I tried to send using it's data and it succeed. So the data is formatted in good manner, just not decoded into the field itself. 

the protobuf file example
```
message Response {
    uint32 sessionId = 1;
    uint32 requestId = 2;
    oneof message {
        GetContextValues getContextValues = 10;
        SetWifiState setWifiState = 11;
        SetWifiCreds setWifiCreds = 12;
        GetWifiNetworkInfo getWifiNetworkInfo = 13;
    }

    enum ProcessingStatus {
        OK = 0;
        Unsupported = 1;
        ServiceError = 2;
        GeneralError = 3;
    }

    message GetContextValues {
       ...
    }

    message SetWifiState {
        ProcessingStatus status = 1;
    }

    message SetWifiCreds {
        ProcessingStatus status = 1;
    }
   ...
```

The message has some uint32 fields and one oneof { with some nested submessages }. I set SetWifiState to the Response.message.setWifiState, Response.which_message = Response_setWifiState_tag, encoded it to data and sent it to mobile. At mobile size I decoded it into Response - both fields sessionId and requestId were filled, but Response.message was nil.

I checked what actually happened. The oneof field is encoded with fieldNumber number == 3, which I can say next by requestId field number == 2. I try to decode this field and it contains uint32 data == 11 that is same value I was set to which_message property of Response.

it looks like some logic was not implemented on the swift protobuf side.

I tried to also option (nanopb_msgopt).no_unions = true; to make this messages to be generated not a part of union but like optionals. I set has_setWifiState = true and setWifiState in the response. But in this case there is no data of  sub message on the mobile side at all, even unknownField is empty.

PetteriAimonen, 2021-02-25:
Yes, nanopb is supposed to interoperate with any other protobuf library.

Can you take hex dump of the data that was transferred?
You can then check what it contains using this tool: https://protogen.marcgravell.com/decode

truebucha, 2021-02-25:
I print the data in base64 string
`CE0QiAEYCw==`
And checked it in the tool 
```
Field #1: 08 Varint Value = 77, Hex = 4D
Field #2: 10 Varint Value = 28, Hex = 1C
Field #3: 18 Varint Value = 11, Hex = 0B
```
I also printed the object from swift:
```
▿ client.Response:
sessionId: 77
requestId: 136
3: 11

  - sessionID : 77
  - requestID : 136
  - message : nil
  ▿ unknownFields : UnknownStorage
    ▿ data : 2 bytes
      - count : 2
      ▿ pointer : 0x000000016f78d410
        - pointerValue : 6165156880
      ▿ bytes : 2 elements
        - 0 : 24
        - 1 : 11
```

truebucha, 2021-02-25:
Found the issue. This is the code that encodes values:
```
    status = pb_encode(&stream, Response_SetWifiState_fields, &response); // instead of Response_fields
```
It was working partially even with that bug.

closed 2021-02-25 by truebucha