nanopb/nanopb #643: execution stuck when init_default large message

zorce, 2021-03-04:
Hi,
Currently have a problem on a microcontroller when trying to initialize a rather large message.
```
#define Request_size 469
#define Oneof_message_size 258
Request req = Request_init_default;
```

The Request message is just holds 8 oneof messages. One of these messages I try to add just one more optional field. Then I run into some issue with the initialization of the message. The above sizes works just fine, but adding one more optional field I get the following sizes:
```
#define Request_size 469
#define Oneof_message_size 265
Request req = Request_init_default; <-- FAILS
```
[image]

Any ideas on this?

PetteriAimonen, 2021-03-04:
Maybe you are running out of stack space? Check the value of stack pointer, and the start of the allocated stack area.

zorce, 2021-03-05:
Yea, tried to expand the stack with 1024 but still the same issue. stuck at some RAM_ptr assignment :/

zorce, 2021-03-05:
Thanks for the input! You where right, the stack size was to small. Thanks again!

closed 2021-03-05 by zorce