nanopb/nanopb #649: Array size defined or declared

AABelkhiria, 2021-03-22:
Hello

Since our project requires a lot of data transmission between nodes and devices, we thought about using Nanopb since it saves us transmission time while encoded.
To do so, we decided to have the encode decode process as a third party and not in the main project.
The problem is the 2 projects are not synchronized regarding the array sizes. It means that when we change the array size in the proto file, we need to do it in our project too.

Example:
```
// proto file
  repeated setting_t setting = 1;

// option file
settings_t.setting type:FT_STATIC, max_count:1000
```
that will generate
```
typedef struct _settings_t { 
    pb_size_t point_setting_count;
    setting_t point_setting[1000];
}  settings_t;
```

and in our project we have for instance
```
static settings_t settings;
static point_t point[1000]; ///< we should have as many points as in the settings
```

My request is the following, instead of having `1000`, is there a possibility to have this value as a define or a constant ? so that it could be used in both places and remove the Magic number

PetteriAimonen, 2021-03-22:
Maybe `pb_arraysize` will work for you also, like it did in #640:

    int data[pb_arraysize(MyMessage, my_array)];

AABelkhiria, 2021-03-22:
It's always better to have the value defined but this will work
Thanks

AABelkhiria, 2021-03-22:
In the same scope,
Proto file 1
```
// proto file
  repeated float value = 1;

// option file
*.value type:FT_STATIC, max_count:1000
```

Proto file 2
```
// proto file
  repeated float prev_value = 1;

// option file
*.prev_value type:FT_STATIC, max_count:1000
```

These 2 arrays have the same count. Is there a work in progress that will handle this case in the near future ?
like setting somewhere a variable `point_count = 1000` and use it in both option files ?

PetteriAimonen, 2021-03-22:
It has been suggested before, but I'm not aware of a syntax that would work with protoc.

closed 2021-03-22 by AABelkhiria