nanopb/nanopb #650: I encountered a problem when decoding the continuous message with a repeated submessage.

slightcycle, 2021-03-23:
I created a test .proto with a simple message:
message test_cross_para_t{
    message repeated_data_t{
        required int32 test1 = 1;
        required int32 test2 = 2;
    };
    repeated repeated_data_t data = 1;
    required int32 count = 2;
}.

 In my encoding code, I encoded ten test_cross_para_t structs continuously. When I encoded each test_cross_para_t struct, I encoded ten repeated_data_t structs in repeated "data" with encode-callback. Here is the sample of encoding code:

    uint8_t *buffer;
    pb_ostream_t stream;
    bool status;

    buffer = new uint8_t[2048];
    stream = pb_ostream_from_buffer(buffer, 2048);
    test_cross_para_t_repeated_data_t repeated_data[10];
   int write_count = 0;
    for (int i = 0; i < 10; i++)
    { 
        test_cross_para_t source = test_cross_para_t_init_zero;
        for (int j = 0; j < 10; j++)
        {
            repeated_data[j].test1 = (write_count++);
            repeated_data[j].test2 = (write_count++);
        }
        source.data.funcs.encode = my_encode;
        source.data.arg = &repeated_data;
        source.count = 0x11;
        status = pb_encode(&stream, test_cross_para_t_fields, &source);
        if (!status)
        {
            printf("can't set more para");
            break;
        }
    }

static bool my_encode(pb_ostream_t *stream, const pb_field_t *field, void *const *arg)
{
    test_cross_para_t_repeated_data_t *repeated_data = (test_cross_para_t_repeated_data_t *)(*arg);
    for (int i = 0; i < 10; i++)
    {
        if (!pb_encode_tag_for_field(stream, field))
        {
            return false;
        }
        if (!pb_encode_submessage(stream, test_cross_para_t_repeated_data_t_fields, &repeated_data[i]))
        {
            return false;
        }
    }

    return true;
}

When I was using a decode callback to decode what I encoded, I only could get the information of repeated_data_t struct. Here is the decoding code:

    pb_istream_t outstream;
    outstream = pb_istream_from_buffer(buffer, stream.bytes_written);
    test_cross_para_t_repeated_data_t outrepeated_data[10];
    int data_count = 0;
    bool istatus;
    while (1)
    {
        test_cross_para_t isource = test_cross_para_t_init_zero;
        isource.data.funcs.decode = my_decode;
        isource.data.arg = &outrepeated_data;
        istatus = pb_decode(&outstream, test_cross_para_t_fields, &isource);
        if (!istatus)
        {
            break;
        }
    }

static bool my_decode(pb_istream_t *stream, const pb_field_t *field, void **arg)
{
    test_cross_para_t_repeated_data_t out_repeated_data = test_cross_para_t_repeated_data_t_init_default;
    {
        if (!pb_decode(stream, test_cross_para_t_repeated_data_t_fields, &out_repeated_data))
        {
            return false;
        }
    }

    return true;
}
 I added some printing information throughout the funcitons about decoding, it seems that the decoding process will never stop unless all the bytes in the encoded buffer are decoded after the decode-callback was triggered. But my purpose is to get the information of one "count" and ten "data" at a time, so I can get all information after 10 cycles.
 I have tried to add some code in the pb_decode_inner in the pb_decode.c, like this:
        if (!decode_field(stream, wire_type, &iter))
            return false;
       // What I added
        if (tag == fields->largest_tag)
        {
            break;
        }
It seems that is works correctly. I don't know if it is a right way to avoid this kind of problem.
 Thanks a lot!

PetteriAimonen, 2021-03-23:
Protocol buffers messages by themselves have no indication of their end. If you concatenate two messages back-to-back, they will get combined. This is required behavior by the standard.

However you can use `pb_encode_delimited()` and `pb_decode_delimited()` which add extra information about the end of each message, and stop decoding after the message ends even if the stream continues.

slightcycle, 2021-03-23:
I'll have a try what you suggested. Thanks for your guidance.

closed 2022-05-30 by PetteriAimonen