nanopb/nanopb #651: Add support for removing deprecated fields from max_size #defines

shreyasbharath, 2021-03-23:
It'd be great to exclude deprecated fields from the max_size #defines in the `pb.h` files. This will decrease RAM usage when there's no intention to use the deprecated fields anyway.

This could be a compiler option so as to not break existing behaviour?

PetteriAimonen, 2021-03-23:
Maybe just add `(nanopb).type = FT_IGNORE` on the deprecated fields, if they are not used in C code.

shreyasbharath, 2021-03-23:
Didn't know that @PetteriAimonen , thanks for the fast response! Should have RTFM 😆

shreyasbharath, 2021-03-23:
Is it possible to add two properties like so?

```
sint32 blah = 6 [deprecated = true, (nanopb).type = FT_IGNORE];
```

PetteriAimonen, 2021-03-23:
Sure, should work. You can also put nanopb-specific options in separate `.options` file if preferred.

closed 2022-05-30 by PetteriAimonen