nanopb/nanopb #653: CMake install failure

samthesuperhero, 2021-04-01:
Hi, I'm using CLion and CMake,  

it recognized two targets, py and static,  

I could manage to build static, but when I install I see this:

```
====================[ Install | Debug ]=========================================
F:\CLion\bin\cmake\win\bin\cmake.exe --build F:\dev\c\nanopb\cmake-build-debug --target install -- -j 6
[ 20%] Generating nanopb_pb2.py
[100%] Built target protobuf-nanopb-static
process_begin: CreateProcess(NULL, protoc --python_out=F:/dev/c/nanopb/cmake-build-debug -IF:/dev/c/nanopb/generator/proto F:/dev/c/nanopb/generator/proto/nanopb.proto, ...) failed.
make (e=2): Не удается найти указанный файл.
mingw32-make.exe[2]: *** [CMakeFiles\generate_nanopb_pb2.py.dir\build.make:80: nanopb_pb2.py] Error 2
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:97: CMakeFiles/generate_nanopb_pb2.py.dir/all] Error 2
mingw32-make.exe: *** [Makefile:149: all] Error 2
```  

At py - it even did not build it at all.

samthesuperhero, 2021-04-02:
> Sounds like it is not finding `protoc.exe`.

@PetteriAimonen - then howe can I find it?

mathisloge, 2021-05-31:
https://github.com/nanopb/nanopb/blob/29b8ee492ec33dbd6f2f25203705803cb70b28e4/CMakeLists.txt#L17 set `nanopb_PROTOC_PATH` to your executable when configuring cmake

closed 2022-05-30 by PetteriAimonen