nanopb/nanopb #655: CMake failed to install '_utils.py'

SergeySeroshtan, 2021-04-09:
## Steps to reproduce
1. Configure project with option `-Dnanopb_BUILD_GENERATOR=OFF`
2. Run `make install`

## Actual behaviour
Got error message like
```sh
CMake Error at _deps/nanopb-build/cmake_install.cmake:49 (file):
  file cannot create directory: /proto.  Maybe need administrative
  privileges.
```

## The reason

CMake variable `PYTHON_INSTDIR` is not defined because option `nanopb_BUILD_GENERATOR ` is `OFF`.
As the result next command is broken:
```cmake
install( FILES generator/proto/_utils.py
         DESTINATION ${PYTHON_INSTDIR}/proto/ )
```

## Expected behaviour
- File `generator/proto/_utils.py` might not be installed when option `nanopb_BUILD_GENERATOR ` is `OFF`
- Or define variable `PYTHON_INSTDIR` outside `if(nanopb_BUILD_GENERATOR)` condition.

PetteriAimonen, 2021-04-09:
Sounds like this might be related to #556.

@dflogeras Do you want to take a look?

SergeySeroshtan, 2021-04-09:
Yep.
Write here https://github.com/nanopb/nanopb/pull/556/files#diff-1e7de1ae2d059d21e1dd75d5812d5a34b0222cef273b7c3a2af62eb747f9d20aR69

dflogeras, 2021-04-10:
I can't seem to reproduce this with master, and it looks like it may have been addressed already with #652 

@SergeySeroshtan  Were you testing on a specific version?  Could you retry with master?

SergeySeroshtan, 2021-04-13:
I was testing version 0.4.5.
This issue is fixed in the `master`.
Thanks.

PetteriAimonen, 2021-04-13:
Thanks for looking into it!

I'll leave the issue open until 0.4.6 is published in case someone else hits it also.

closed 2022-05-30 by PetteriAimonen