nanopb/nanopb #656: Problem with python venv on Windows

kalj, 2021-04-12:
Currently, nanopb_generate doesn't correctly handle a python venv. Basically, when doing this:
```cmd
python -m venv myvenv
call myvenv\Scripts\activate.bat
protoc.exe --plugin=protoc-gen-nanopb=path\to\protoc-gen-nanopb.bat myfile.proto
```
I expect it to use `myenv\Scripts\python.exe` (which is the first match in the PATH). However, the system-wide default is used instead.
The reason for this is that `protoc-gen-nanopb.bat` uses the `py.exe` launcher which wants to decide on its own which python executable it should use. It is wellknown that `py.exe` has problems with virtual environments (https://bugs.python.org/issue38999). The following change seems to make it work:

```diff
diff --git a/generator/protoc-gen-nanopb.bat b/generator/protoc-gen-nanopb.bat
index 48a4aa9..fa5bdd2 100644
--- a/generator/protoc-gen-nanopb.bat
+++ b/generator/protoc-gen-nanopb.bat
@@ -9,4 +9,4 @@
 :: --plugin= on the command line.
 set mydir=%~dp0
-py -3 "%mydir%\nanopb_generator.py" --protoc-plugin
+python "%mydir%\nanopb_generator.py" --protoc-plugin %*
```

If this results in python not being found on some systems, I suggest adding some way to allowing the user to specify which python executable to use, instead of relying on the problematic `py.exe`.

PetteriAimonen, 2021-04-12:
What a mess. On systems without Python in PATH, py.exe seemed to be more reliable.

I guess feel free to pull request this if it works for you.

kalj, 2021-04-13:
Solved with #657

closed 2021-04-13 by kalj