nanopb/nanopb #658: Guidance required: GRPC over GSM modem (simcom 800 c) using AT commands

supersaeyan, 2021-04-13:
Sorry for posting this as an issue, couldn't find useful information on reddit.
Please feel free to close this, but any related resources would be really appreciated.

I wanted to implement GRPC on STM32 based microcontroller with sim 800 c, because of the payload size difference as compared to HTTP 1.1, to lower GSM tariffs.

I was going through an example https://github.com/nanopb/nanopb/blob/master/examples/simple/simple.c

At Line 42, there's a comment explaining the usage

`    /* Now we could transmit the message over network, store it in a file or
     * wrap it to a pigeon's leg.
     */`

My question is, if i write the encoded data a binary payload using GSM modem AT commands, wouldn't it be equivalent to writing string representation(hexdigest?) or the binary payload?

P.S I am a backend engineer, barely know anything about embedded, trying to convince everyone in my company to implement protobuf.

PetteriAimonen, 2021-04-13:
Typically for lowest data transfer sizes, you would want to transmit the data as binary to server. I haven't checked SIM800C AT commands to see how that is done, but typically there is either a way to specify the length of the data sent to the modem or a way to indicate its end.

The network server example uses plain protobuf bidirectional communication between client and server. There is also some work on grpc in issue #235 but I haven't tried it myself. GRPC typically works over HTTP2 which would add its own overhead.

supersaeyan, 2021-04-13:
@PetteriAimonen

So you recommend opening a TCP socket and that basically implement communication protocol from scratch. That is doable and i agree that it would result in minimal payload size, but i read a few issues about sending binary data over AT commands.
Causes server side scaling issues as the requests become stateful if sessions are persistent for resumability.

https://stackoverflow.com/questions/59859070/how-to-use-nanopb-over-gsm-modem-communication-where-specific-characters-like-ct 

But thanks for the binary data approach, i will look into it a bit more. But just for a reference, any figures on how much overhead http 1.1 and http 2 add, compared to binary?

PetteriAimonen, 2021-04-13:
Transferring the binary data with HTTP or HTTP2 encapsulation will have the same requirements for escaping special characters. It just something one has to deal with when using AT command modems. You can hex-encode payload but it will double the size.

HTTP overhead depends a lot on how you implement it, but typically it would be around 50-500 bytes per request.

supersaeyan, 2021-04-13:
So apart from the asymmetric communication part that reduces unnecessary request overhead, how much payload size reduction should be expected?

Also, I have implemented bit torrent protocol and used struct packing for binary payloads. Is that the best way, the only way?

PetteriAimonen, 2021-04-13:
That will depend on so many factors that you'll just have to make your own tests.

supersaeyan, 2021-04-13:
I should get to work then.

Would you mind connecting elsewhere?
I feel bad using an issue on this repo for something which is now unrelated(sort of).

PetteriAimonen, 2021-04-13:
I'm not really prepared to provide that much support for free, but for commercial support you can contact me through https://www.devembedded.com/en/services/consulting.html

supersaeyan, 2021-04-13:
And here i thought we were friends :smiley: 

Thanks for the help though @PetteriAimonen

closed 2022-05-30 by PetteriAimonen