nanopb/nanopb #659: CMake nanopb_generate_cpp does not find *.options file

fermentedfly, 2021-04-19:
Hi, I'm trying to use the generator via CMake with a options file and the options file is not used / found by nanopb_generator.py

The issue seems to be resolved when the proto file is passes as a relative path to the current directory

### Steps to repoduce

Add a simple.options to examples/cmake_simple containing
```
SimpleMessage.lucky_number int_size:IS_8
```

Add verbose logging to CMakeLists.txt (doesn't matter, just makes the issue easier to trace)
```
set(NANOPB_OPTIONS -v)
```
When running make, nanopb_generator.py produces the following
```
Nanopb version nanopb-0.4.6-dev
Google Python protobuf library imported from /usr/lib/python3.9/site-packages/google/protobuf/__init__.py, version 3.15.8
Options file not found: simple.options
Options for simple.proto: 
...
```

Changing nanopb_generate_cpp in CMakeLists.txt to
```
nanopb_generate_cpp(PROTO_SRCS PROTO_HDRS ./simple.proto)
```
Resolved the issue

NZSmartie, 2021-04-29:
I'm also experiencing a similar issue. Zephyr project recently added nanopb support in https://github.com/zephyrproject-rtos/zephyr/commit/b6979736ca61656c965fbacdcb46c8b6c39adada

In my `CMakeLists.txt` I have tried
```cmake
# Test using relative to current directory
nanopb_generate_cpp(proto_sources proto_headers RELPATH .
    proto/test.proto
)

# Test using relative to the source directory
nanopb_generate_cpp(proto_sources proto_headers RELPATH ${CMAKE_SOURCE_DIR}
    proto/test.proto
)

# Test using relative to the proto directory (Does not compile in zephyr with "arm-zephyr-eabi-gcc: error: test.pb.c: No such file or directory")
nanopb_generate_cpp(proto_sources proto_headers RELPATH proto
    proto/test.proto
)
```

Doing a bit of debugging, the `NANOPB_GENERATE_CPP` function in `FindNanopb.cmake` does  detect the `.options` file adjacent to the `.proto` file and adds the directory to `NANOPB_OPTIONS_DIRS`. The resulting generated command for the build system to execute looks like

```ninja
#############################################
# Custom command for proto/test.pb.c

build proto/test.pb.c proto/test.pb.h: CUSTOM_COMMAND ../proto/test.proto nanopb/generator/proto/nanopb_pb2.py ../proto/test.options || etl zephyr/driver_validation_h_target zephyr/kobj_types_h_target zephyr/offsets zephyr/offsets_h zephyr/parse_syscalls_target zephyr/syscall_list_h_target zephyr/zephyr_generated_headers
  COMMAND = cd /home/nzsmartie/Projects/test/build && /home/nzsmartie/.local/bin/protoc -I/home/nzsmartie/Projects/test/build/nanopb/generator -I/home/nzsmartie/Projects/test/build/nanopb/generator/proto -I/home/nzsmartie/Projects/test/build -I/home/nzsmartie/Projects/test --plugin=protoc-gen-nanopb=/home/nzsmartie/Projects/test/build/nanopb/generator/protoc-gen-nanopb "--nanopb_opt= -I/home/nzsmartie/Projects/test/proto" --nanopb_out=/home/nzsmartie/Projects/test/build /home/nzsmartie/Projects/test/proto/test.proto
  DESC = Running C++ protocol buffer compiler using nanopb plugin on proto/test.proto
  restat = 1
```

Looking into how proto files are parsed. 
- It appears that the current directory being searched for `.options` files, is the build output directory, not the source directory. 
- the file name being searched is the **relative** path of the `.proto` file from the source directory
  - The `--nanopb_opt` optin includes the current directory the optins file was found in, not the root directory of the source. So `nanopb_generator.py` will look for
    - `./proto/test.options`
    - `/home/nzsmartie/Projects/test/proto/proto/test.options`
    - `proto/proto/test.options`

https://github.com/nanopb/nanopb/blob/29b8ee492ec33dbd6f2f25203705803cb70b28e4/generator/nanopb_generator.py#L2083-L2099

Either the `nanopb_generator.py` is altered to search the source directory, and/or  `FindNanopb.cmake` is adjusted to add the root of the relative directory to the search list. instead of the current directory the options file was discovered in. 

P.S. my triage might not be accurate, but as an outsider working with nanopb for the first time, this is the best I could figure out.

NZSmartie, 2021-04-29:
A current work around is to add the following to `CMakeLists.txt`
```cmake
set(NANOPB_OPTIONS "-I${CMAKE_SOURCE_DIR}")
```

palchak-google, 2022-04-08:
There is indeed a bug in the FindNanopb.cmake file, as follows:

On line 201 the code extracts the path of the `.proto` file being processed and stores this value in the `FIL_DIR` variable:
https://github.com/nanopb/nanopb/blob/6f7bca0e1042558e3e98f5add764b260a70321f1/extra/FindNanopb.cmake#L198-L201

If the `.proto` file was given as a bare filename, i.e. does not have any path specified (not even `./`), then the `FIL_DIR` variable will be empty. This is important.

Later, the code uses the `FIL_DIR` variable to compose the name of the `.options` file that corresponds to the current `.proto` file:
https://github.com/nanopb/nanopb/blob/6f7bca0e1042558e3e98f5add764b260a70321f1/extra/FindNanopb.cmake#L222-L224

Notice here that if `FIL_DIR` is empty, the resulting value for `ABS_OPT_FIL` will begin with a forward slash. Thus `ABS_OPT_FIL` will refer to a file in the filesystem's root directory, which is definitely wrong.

The fix is to insert the following code between lines 222 and 223:
```
if (NOT FIL_DIR)
    set(FIL_DIR ".")
endif()
```
This code ensures that when `FIL_DIR` is concatenated with `FIL_WE` the result maintains the correct path information.  I have tested this specific change in my project and it works.

Alternatively, one could fix this issue by replacing line 201 with:
```
get_filename_component(FIL_DIR ${ABS_FIL} PATH)
```
This alternate fix ensures that `FIL_DIR` will never be blank by getting the path from the absolute filename and not from the relative filename.

PetteriAimonen, 2022-04-09:
@kfitch @kurddt Do you want to review @palchak-google's suggestion above?

I think changing `FIL_DIR` to absolute on line 201 makes more sense, as it is only used to form `ABS_OPT_FIL` which is absolute path anyway.

closed 2022-05-30 by PetteriAimonen