nanopb/nanopb #660: array max size exceeded

c0diq, 2021-04-27:
I have a very simple proto with a `repeated` field.

```
syntax = "proto2";

message Info {
    required int32 value = 1;
}

message Update {
    repeated Info info = 1;
}
```

Using the callback mechanism, I'm getting the `array max size exceeded` encoding failed.

```

typedef struct _ValueInfo {
    int value;
} ValueInfo;

typedef struct _pb_callback_info_parameter_s {
    int count;
    ValueInfo values[10];
} pb_callback_info_parameter_s;

bool callback_encode_values(pb_ostream_t *ostream, const pb_field_t *field, void *const *arg) {
    bool is_success = true;

    if ((NULL == arg) || (NULL == *arg)) {
        is_success = false;
    } else {
        const pb_callback_info_parameter_s *callback_parameter = (const pb_callback_info_parameter_s *)(*arg);
        ValueInfo *values = (ValueInfo*)callback_parameter->values;
        size_t values_count = callback_parameter->count;

        for (size_t count = 0; count < values_count; count++) {
            Info info = Info_init_zero;
            info.value = values[count].value;

            /* This encodes the header for the field, based on the constant info from pb_field_t. */
            if (!pb_encode_tag_for_field(ostream, field)) {
                is_success = false;
            }

            /* This encodes the data for the field, based on our Tire structure. */
            if (!pb_encode_submessage(ostream, Info_fields, &info)) {
                is_success = false;
            }
        }
    }

    return is_success;
}

int main(int argc, const char * argv[]) {
    if (argc < 2) {
        fprintf(stderr, "Missing file path to save data to.");
        return 1;
    }

    FILE *fd = fopen(argv[1], "wb+");
    if (fd == NULL) {
        fprintf(stderr, "Failed to open output file");
        return 1;
    }

    Update update = Update_init_zero;
    pb_callback_info_parameter_s param = { 1, { 1 } };
    update.info.arg = (void*)(&param);
    update.info.funcs.encode = &callback_encode_values;

    pb_ostream_t output = pb_ostream_from_file(fd);
    if (!pb_encode(&output, Update_fields, &update)) {
        fprintf(stderr, "Encoding failed: %s\n", PB_GET_ERROR(&output));
        goto fail;
    }

    fclose(fd);
    return 0;

fail:
    fclose(fd);
    return 1;
}
```

PetteriAimonen, 2021-04-28:
The error message `array max size exceeded` would mean that a non-callback array is given a size that is larger than the `max_size` that has been defined in a `.proto` or `.options` file.

However, there isn't such field in your example. Are you sure that the example matches the code that you are getting the error with, and that all files have been regenerated and recompiled?

c0diq, 2021-04-28:
Yes I have no .options file and the protocol is exactly what I use.

PetteriAimonen, 2021-04-28:
Seems to work for me:
https://jpa.kapsi.fi/stuff/other/issue_660.tar.gz

    petteri@raichu:~/nanopb/issue_660$ ./test | protoc --decode=Update test.proto
    info {
      value: 1
    }

PetteriAimonen, 2021-05-01:
gcc -Wall -o test test.c test.pb.c -I.. ../pb_encode.c ../pb_common.c

c0diq, 2021-05-05:
It worked for me with gcc and your command line however it fails with Xcode. Could clang vs gcc settings make it fail?

PetteriAimonen, 2021-05-06:
It builds and works the same for me on clang 10:

<pre>
petteri@raichu:~/nanopb/issue_660$ clang --version
clang version 10.0.0-4ubuntu1 
Target: x86_64-pc-linux-gnu
petteri@raichu:~/nanopb/issue_660$ clang -Wall -o test test.c test.pb.c -I.. ../pb_encode.c ../pb_common.c
test.c:43:49: warning: suggest braces around initialization of subobject [-Wmissing-braces]
    pb_callback_info_parameter_s param = { 1, { 1 } };
                                                ^
                                                {}
1 warning generated.
petteri@raichu:~/nanopb/issue_660$ ./test | protoc --decode=Update test.proto
info {
  value: 1
}
</pre>

At this point this sounds like a problem with your build environment rather than a problem with nanopb.

You could try stepping through in a debugger for both the working binary and the Xcode-built one, and see where they work differently.

c0diq, 2021-05-06:
Thanks. I've filed a bug with the spm team.

closed 2022-05-30 by PetteriAimonen