nanopb/nanopb #661: Default value of oneof

AABelkhiria, 2021-04-29:
Hello,

I am writing this Issue to ask about a how to.
we want to ask is we're able to set a default value for a which oneof
```
message msg_structure {
  oneof version {
    uint32 v1 = 1;
    uint32 v2 = 2;
  }
}
```

The goal is to set `which_version` in `msg_structure_init_default` to `msg_structure_v2_tag`

PetteriAimonen, 2021-04-29:
No, currently there is no support for that.

AABelkhiria, 2021-04-29:
Thanks for your quick reply.

Is this feature planned in NanoPb roadmap or is it something related to protoc ?

PetteriAimonen, 2021-04-29:
It could possibly be implemented as a nanopb-specific generator option, but I'm not sure if there is enough use for it.

PetteriAimonen, 2022-05-30:
There are currently no plans to implement this.

closed 2022-05-30 by PetteriAimonen