nanopb/nanopb #662: Proto3 float wrong wire type

Gator96100, 2021-05-04:
I’m using nanopb for a serial communication between a pc and a STM32 running FreeRTOS and I did run into a problem with floats when using proto3, all other types work. Unfortunately, I couldn’t replicate the issue without the STM32. The STM32 sends messages to the pc and I have a small program that listens and decodes all incoming messages. When I decode a message that has a float variable, I get the error `wrong wire type`, but when I specify `(nanopb).proto3 = false` everything works. 

messages.proto (non working):
```
syntax = "proto3";
import 'nanopb.proto';

package gator96100.work;

message Msg1 {
    float f1 = 1;
}

message Message {
    oneof payload {
		Msg1 m1 = 1;
    }
}
```
messages.proto (working):
```
syntax = "proto3";
import 'nanopb.proto';

package gator96100.work;

message Msg1 {
    float f1 = 1 [(nanopb).proto3 = false];
}

message Message {
    oneof payload {
		Msg1 m1 = 1;
    }
}

```
listener:
```
#include <stdio.h>   /* Standard input/output definitions */
#include <string.h>  /* String function definitions */
#include <unistd.h>  /* UNIX standard function definitions */
#include <fcntl.h>   /* File control definitions */
#include <errno.h>   /* Error number definitions */
#include <termios.h> /* POSIX terminal control definitions */
#include <pb_encode.h>
#include <pb_decode.h>
#include "messages.pb.h"

int set_interface_attribs (int fd, int speed, int parity){
        struct termios tty;
        if (tcgetattr (fd, &tty) != 0)
        {
                printf("error %d from tcgetattr\n", errno);
                return -1;
        }

        cfsetospeed (&tty, speed);
        cfsetispeed (&tty, speed);

        tty.c_cflag = (tty.c_cflag & ~CSIZE) | CS8;     // 8-bit chars
        // disable IGNBRK for mismatched speed tests; otherwise receive break
        // as \000 chars
        tty.c_iflag &= ~IGNBRK;         // disable break processing
        tty.c_lflag = 0;                // no signaling chars, no echo,
                                        // no canonical processing
        tty.c_oflag = 0;                // no remapping, no delays
        tty.c_cc[VMIN]  = 0;            // read doesn't block
        tty.c_cc[VTIME] = 5;            // 0.5 seconds read timeout

        tty.c_iflag &= ~(IXON | IXOFF | IXANY); // shut off xon/xoff ctrl

        tty.c_cflag |= (CLOCAL | CREAD);// ignore modem controls,
                                        // enable reading
        tty.c_cflag &= ~(PARENB | PARODD);      // shut off parity
        tty.c_cflag |= parity;
        tty.c_cflag &= ~CSTOPB;
        tty.c_cflag &= ~CRTSCTS;

        if (tcsetattr (fd, TCSANOW, &tty) != 0)
        {
                printf("error %d from tcsetattr\n", errno);
                return -1;
        }
        return 0;
}

void set_blocking (int fd, int should_block){
        struct termios tty;
        memset (&tty, 0, sizeof tty);
        if (tcgetattr (fd, &tty) != 0)
        {
                printf("error %d from tggetattr\n", errno);
                return;
        }

        tty.c_cc[VMIN]  = should_block ? 1 : 0;
        tty.c_cc[VTIME] = 5;            // 0.5 seconds read timeout

        if (tcsetattr (fd, TCSANOW, &tty) != 0)
                printf("error %d setting term attributes\n", errno);
}

static bool read_callback(pb_istream_t *stream, uint8_t *buf, size_t count)
{
    int fd = (intptr_t)stream->state;
    int result;
    
    if (count == 0)
        return true;

    result = read(fd, buf, count);
    
    if (result == 0)
        stream->bytes_left = 0; /* EOF */
    
    return result == count;
}

pb_istream_t pb_istream_from_socket(int fd)
{
    pb_istream_t stream = {&read_callback, (void*)(intptr_t)fd, SIZE_MAX};
    return stream;
}

int main(int argc, char **argv)
{
	char *portname = "/dev/ttyS14";
	int fd = open (portname, O_RDWR | O_NOCTTY | O_SYNC);
	if (fd < 0){
        printf("error %d opening %s: %s\n", errno, portname, strerror (errno));
        return 1;
	}

	set_interface_attribs (fd, B115200, 0);  // set speed to 115,200 bps, 8n1 (no parity)
	set_blocking (fd, 0);                // set no blocking

	gator96100_work_Message m = gator96100_work_Message_init_default;
	
    pb_istream_t istream = pb_istream_from_socket(fd);
    while(1){
		if (pb_decode_delimited(&istream, gator96100_work_Message_fields, &m)){
			printf("which_payload %d\n",m.which_payload);
	    }else{
		    printf("Decoding failed: %s\n", PB_GET_ERROR(&istream));
	    }
    }	
	return 0;
}
```

PetteriAimonen, 2021-05-04:
Can you add the following information:

- Nanopb version in use
- Hex dump of the raw message data that gets passed to `pb_decode()` when it fails

PetteriAimonen, 2021-05-04:
And maybe also the encoding code, if possible.

Gator96100, 2021-05-04:
I’m using Nanopb 0.4.5.
The issue seems to be more complicated than I thought. When using proto2 the message gets transmitted, but the float value gets removed. I think it has to do with the has_temperature variable (temperature is the value I want to send). 
[image]

encoding (STM32), serial buffer that is send over USB:
proto2 "working" float(123.456f): {0x02 0x5a 0x0}
proto3 non working float(123.456f): { 0x7 0x5a 0x5 0x0d 0x4e 0x22 0xf7 0x42 }
proto3 working int32 (123): {0x4 0x5a 0x2 0x8 0x7b}

decoding (PC), serial buffer coming from read():
proto2 "working" float(123.456f): {0x02 0x5A 0x00}
proto3 non working float: {0x07 0x5A 0x05 0x0A 0x4E 0x22 0xF7 0x42}
proto3 working int32 (123): {0x04 0x5A 0x02 0x08 0x7B}

Somewhere along the transmission of my float the 4th byte gets modified, that’s the reason why I get the error, but it’s to constant to be a random bitflip and only when I try to send a float. So I guess proto3 works fine, but I’m not sure why proto2 does not send the float.

PetteriAimonen, 2021-05-04:
`0x0D` vs. `0x0A` is the CR vs LF line-ending conversion. This usually happens when the serial port is set to ASCII mode instead of binary mode. According to [this tutorial](https://blog.mbedded.ninja/programming/operating-systems/linux/linux-serial-ports-using-c-cpp/) you have to clear `OPOST` flag.

Proto2 optional fields have a separate `has_field` that you must set to true to mark the field as present. In proto3 fields are present whenever they are non-zero.

Gator96100, 2021-05-04:
Thank you very much!
After setting the correct `c_iflag` everything works.

closed 2021-05-04 by Gator96100