nanopb/nanopb #663: HardFault Handler state entered when encoding uint64

AABelkhiria, 2021-05-04:
Hello,

I am using
- `nanopb 0.4.5`
- `libprotoc 3.6.1`
- mcu architecture `ARM Cortex M4`
and trying to encode uint64_t data but the MCU enters a HardFault state.
```
message msg_v1 {
//
  required uint64 id = 3; 
}
```

When I remove uint64 and change it by uint32, the serialization works fine.

PetteriAimonen, 2021-05-04:
What is the value of `SCB->CFSR` status register? It indicates the cause of the hardfault.

Also check the backtrace to see on what line it faults. In most debuggers you can just use the built-in backtrace tool. If you can't use a debugger, add a print to your hardfault handler that takes the 6th stack entry. E.g. [this handler for PSP stack](https://github.com/devEmbedded/decurrent-sw/blob/master/src/debug.c#L90) or modify for MSP stack if that's what the thread is using. Then look into map file to see what function and line that program counter value points to.

AABelkhiria, 2021-08-18:
These are the values of the registers on hardfault
```
[0000015542][MPU][CRITIC]: HardFault:
[0000015543][MPU][CRITIC]: SCB->CFSR   0x01000000
[0000015544][MPU][CRITIC]: SCB->HFSR   0x40000000
[0000015546][MPU][CRITIC]: SCB->MMFAR  0xe000ed34
[0000015547][MPU][CRITIC]: SCB->BFAR   0xe000ed38
[0000015549][MPU][CRITIC]: 
[0000015550][MPU][CRITIC]: SP          0x2000cfb0
[0000015551][MPU][CRITIC]: R0          0x2000d0dc
[0000015553][MPU][CRITIC]: R1          0x2000d058
[0000015554][MPU][CRITIC]: R2          0x00000001
[0000015556][MPU][CRITIC]: R3          0x2000db55
[0000015557][MPU][CRITIC]: R12         0x00000000
[0000015559][MPU][CRITIC]: LR          0x00026125
[0000015560][MPU][CRITIC]: PC          0x00025c00
[0000015562][MPU][CRITIC]: PSR         0x61000200
```

AABelkhiria, 2021-08-18:
[image]
This is where the code breaks.
According to the debugger the value is correct but when casting it to uint64_t it brings a HardFault
- This test is done with
```
required fixed64 id = 3;
```

PetteriAimonen, 2021-08-18:
The SCB->CFSR indicated error is "unaligned access". On Cortex-M4, unaligned 32-bit accesses are ok, but depending on what instruction compiler uses, unaligned 64-bit accesses may or may not be ok.

Is your message structure aligned to word boundary? Check that the address of the structure passed to `pb_encode()` is divisible by 4. If it isn't, check how it is allocated - any standards-compliant memory allocation should allocate it as aligned.

If alignment of the whole structure is not the issue, check whether you are using the `(nanopb).packed_struct = true` option, which could also cause unaligned accesses.

AABelkhiria, 2021-08-25:
It's an alignment issue.
setting packet_struct to `false` fixed the problem

closed 2021-08-25 by AABelkhiria