nanopb/nanopb #664: "Stream Full" error when stream->bytes_written == MAX_SIZE

shanusmcshan, 2021-05-04:
Platform Promicro AVR 32U4 - 

The condition in bold becomes true, causing the return of "stream full" error when stream->bytes_written is 65535 and a subsequent write takes place.  

in pb_encode.c :

bool checkreturn pb_write(pb_ostream_t *stream, const pb_byte_t *buf, size_t count)
{
    if (count > 0 && stream->callback != NULL)
    {
        if (**stream->bytes_written + count < stream->bytes_written** ||
            stream->bytes_written + count > stream->max_size)
        {
            PB_RETURN_ERROR(stream, "stream full");
        }
...

PetteriAimonen, 2021-05-05:
Yeah, I guess that can be an issue when writing long messages using callbacks on a platform with 16-bit `size_t`.

As a workaround you can change the `size_t` to `uint32_t` in `pb_encode.h`.

shanusmcshan, 2021-05-05:
This issue is not caused by writing long messages. The messages are only a few bytes.

bytes_written constantly increases after each message, it only wraps by virtue of the overflow.  So if by chance the message bytes approaches 65535 and adding the count causes the total to wrap then this happens.

PetteriAimonen, 2021-05-05:
Well, if length of a single message is not over 64kB, then you can just reset `bytes_written` before calling `pb_encode()`.

shanusmcshan, 2021-05-05:
Yes that is certainly a possible workaround, as is using a new stream object for each new encode.  But I spent a few hours trying to work out why I would get seemingly  _random_ "stream full" messages.  Feels like a candidate for explicit documentation or a change to the API - at least to make it a bit more obvious?

PetteriAimonen, 2021-05-05:
Yeah, there is certainly a lot to improve in the documentation, and I'm sorry for your lost time.

shanusmcshan, 2021-05-11:
No problem, and thank you for your quick responses