nanopb/nanopb #666: Make doesn't regenerate C files when .options file change

FabianInostroza, 2021-05-11:
Go to network_server example folder and run `make`, then `touch fileproto.options` and `make` again.
Files are not regenerated.

PetteriAimonen, 2021-05-11:
Yeah, the example makefile should be fixed to note the existence of `.options` if it exists.

FabianInostroza, 2021-05-11:
> Yeah, the example makefile should be fixed to note the existence of `.options` if it exists.

How is that done?
In my ignorance about makefiles I tried a lot of things but the change that worked best was in https://github.com/nanopb/nanopb/blob/29b8ee492ec33dbd6f2f25203705803cb70b28e4/extra/nanopb.mk#L35
changing it to:
`%.pb.c %.pb.h: %.proto %.options`

PetteriAimonen, 2021-05-11:
Yeah, something like that.

Not sure if there is a good way to make it conditionally dependent, i.e. make it work also if there is no options file.

FabianInostroza, 2021-05-13:
This rules seems to work fine:
```
.SECONDEXPANSION:
# Rule for building .pb.c and .pb.h
%.pb.c %.pb.h: %.proto $$(wildcard %.options)
	$(PROTOC) $(PROTOC_OPTS) --nanopb_out=. $<
```

I've tried in a project with various .proto files and some of them have associated .options files.

PetteriAimonen, 2022-05-01:
I ended up duplicating the rule because I'm not sure how portable `.SECONDEXPANSION` is. Seems to work now.

closed 2022-05-30 by PetteriAimonen