nanopb/nanopb #667: Xcode 12.5 analyzer shows two warnings in pb_decode.c

paulb777, 2021-05-22:
From https://stackoverflow.com/questions/67644723/updated-firebase-crashlytics-on-ios-getting-a-bunch-of-analyzer-issues

```
/Users/paul/Library/Developer/Xcode/DerivedData/firebase-ios-sdk-ayfzngrufgjzprffkljwxevrtomi/SourcePackages/checkouts/nanopb/pb_decode.c:541:24: warning: Dereference of null pointer [core.NullDereference]
        *(void**)pItem = NULL;
        ~~~~~~~~~~~~~~~^~~~~~
/Users/paul/Library/Developer/Xcode/DerivedData/firebase-ios-sdk-ayfzngrufgjzprffkljwxevrtomi/SourcePackages/checkouts/nanopb/pb_decode.c:547:9: warning: Null pointer passed as 1st argument to memory set function [unix.cstring.NullArg]
        memset(pItem, 0, iter->pos->data_size);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

PetteriAimonen, 2021-05-22:
Hmm, 0.3 branch apparently?

To me this looks like false result by a static analyzer.
The `initialize_pointer_field()` function is called from lines 593, 642 and 670.
Of these 642 and 670 initialize `pItem` right before, so analyzer should be able to see that.
On line 593, the pointer comes from `allocate_field()` which does check against NULLs, but perhaps the logic is too complex for the analyzer, possibly because of the cast from `void*` to `void**`.

If you figure out a way to tell the analyzer "this is ok" (maybe adding some kind of unnecessary `if` check?), I can accept a pull request.

paulb777, 2021-05-22:
Thanks for the reply.  Yes - the 0.3 branch.  

I will do a deeper investigation into options to resolve the analyzer complaints, but it may take me a few weeks to get back to this if no one else picks it up.

carlosaguiar, 2022-01-25:
@paulb777 looks like @PetteriAimonen already fix this, right? It would be great to have this fix merge because it's the only 2 warnings I have when running Xcode analyzer (Xcode 13.2.1)

paulb777, 2022-01-25:
Yep the fix was merged into [maintenance_0.3 ](https://github.com/nanopb/nanopb/compare/maintenance_0.3) from https://github.com/nanopb/nanopb/commit/b6c6381949ed3f7101c95c79beee66a6560e2a10

paulb777, 2022-01-25:
Looks like the 0.3.9.8 is the commit before, so it's pending a 0.3.9.9 release.

PetteriAimonen, 2022-01-26:
Yeah, it is not important enough for a new maintenance release.

Eric-Catalyst, 2022-04-20:
> Hmm, 0.3 branch apparently?
>
> To me this looks like false result by a static analyzer. The `initialize_pointer_field()` function is called from lines 593, 642 and 670. Of these 642 and 670 initialize [...]

Actually, if you look in detail at the static analyzer results, it is accounting for the assignments to pItem and the null check, so  the calls on lines 593 and 670 are not at issue. The only issue is the call via line 642. Unfortunately, the assignment on line 641 could result in pItem being null. You are assuming that the data being read in the loop is well formed and valid, which is not a safe assumption.

PetteriAimonen, 2022-04-20:
@Eric-Catalyst @paulb777  Can you elaborate? I will make an updated 0.3 maintenance release if needed, but first I have to understand the issue.

To me it still appears that assignment on line 641 (in 0.3 branch) won't write `NULL` unless `iter->pData` is `NULL`. And that shouldn't be `NULL` when `allocate_field()` on line 633 has succeeded.

(The division of trusted vs. untrusted data may be relevant to discussion: https://jpa.kapsi.fi/nanopb/docs/security.html#division-of-trusted-and-untrusted-data )

Eric-Catalyst, 2022-04-20:
Line 633 is not always called, there is a path to line 641 that bypasses line 633. if you run the static analyzer, you will see the execution path that leads to the error condition.

PetteriAimonen, 2022-04-20:
@Eric-Catalyst 
I don't have Xcode, but reading the source code I cannot identify a condition where line 633 wouldn't be executed at least once before line 641.

Do you have an example the causes actual null pointer dereference when executed?

Eric-Catalyst, 2022-04-20:
Okay, I see what you are doing. I agree that line 633 will always execute in the first pass through the loop, which should prevent a null pointer. 

However, it is still better to have a release that passes the static analyzer as I will have problems getting the code that uses this library through code review with a null pointer warning in static analysis. I suspect I am not the only one. 

Looks like an interesting case for the Static Analysis team at Apple to look at. This is the first false positive I have seen in quite a while.

PetteriAimonen, 2022-04-20:
Ok, thanks for confirming my interpretation :)
Indeed the code is quite convoluted, but for now it seems the most straightforward way to implement the logic.

I'm ok with making a 0.3.9.9 release for this, as the warning indeed seems to be impacting many people.
Can you confirm that the newest code in maintenance_0.3 branch does not trigger any warnings in newest Xcode analysis?
If it does, it would make sense to fix them at the same time.

Eric-Catalyst, 2022-04-20:
It is going to take me a bit of time to figure out how to best test this - it is an indirect dependency in my projects....

paulb777, 2022-04-20:
Confirmed that the tip of maintenance_0.3 at b6c6381949ed3f7101c95c79beee66a6560e2a10 builds cleanly in the Xcode 13.2.1 analyzer

Eric-Catalyst, 2022-04-20:
I don't see any static analysis issues.

You should be good to release a 0.3.9.9 release.

Have you considered moving to a more traditional release numbering? Google seems to have issues with your numbering.

PetteriAimonen, 2022-04-21:
Yes, I have been considering making 0.5.x series eventually 1.x and moving to semantic versioning.

I'll make the 0.3.9.9 release this week.

PetteriAimonen, 2022-04-23:
0.3.9.9 is now published.
I included also a fix for some integer overflow sanitizer warnings in newer clang versions.

closed 2022-04-23 by PetteriAimonen

Eric-Catalyst, 2022-05-02:
> ** EMAIL ORIGINATED FROM OUTSIDE THE COMPANY. PLEASE USE DISCRETION IF OPENING ATTACHMENTS OR CLICKING LINKS **

Thank you!

[signature_1472570716]
Eric Hildum | Senior Principal
m: +1 510-305-0801
catalyst.concentrix.com<http://catalyst.concentrix.com/>

This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to which they are addressed.