nanopb/nanopb #668: Several warnings about uninitialized variables in FindNanopb.cmake

kalj, 2021-05-27:
When using `--warn-uninitialized` in a project using cmake with Nanopb, many warnings from inside `extra/FindNanopb.cmake` arise, e.g.:
```
CMake Warning (dev) at externals/nanopb/external/extra/FindNanopb.cmake:342 (find_program):
  uninitialized variable 'PROTOBUF_SRC_ROOT_FOLDER'
Call Stack (most recent call first):
  externals/nanopb/CMakeLists.txt:27 (find_package)
```
and:
```
CMake Warning (dev) at externals/nanopb/external/extra/FindNanopb.cmake:275 (add_custom_command):
  uninitialized variable 'NANOPB_DEPENDS'
Call Stack (most recent call first):
  proto/CMakeLists.txt:8 (nanopb_generate_cpp)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at externals/nanopb/external/extra/FindNanopb.cmake:292 (set):
  uninitialized variable 'NANOPB_SRCS'
Call Stack (most recent call first):
  proto/CMakeLists.txt:8 (nanopb_generate_cpp)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at externals/nanopb/external/extra/FindNanopb.cmake:293 (set):
  uninitialized variable 'NANOPB_HDRS'
Call Stack (most recent call first):
  proto/CMakeLists.txt:8 (nanopb_generate_cpp)
```

I tried adding e.g.
```
if(NOT DEFINED NANOPB_DEPENDS)
set(NANOPB_DEPENDS )
endif()
```
But I was not able to get everything away, and there might be better ways to solve the underlying problem.

peter-toft-greve, 2021-05-27:
Thanx for filing @kalj

PetteriAimonen, 2022-05-30:
Note about reproducing this problem: `cmake --warn-uninitialized` only warns about files that are under the project root folder. To test this with e.g. `examples/cmake_simple`, `--check-system-vars` has to be given also.