nanopb/nanopb #670: Support decoding in multiple chunks

yonsch, 2021-06-01:
**What should the feature do?**
`pb_decode` receives an entire message at once, continuous in memory. It would be useful to have a `pb_decode_partial` function that receives a buffer and a byte count. It will read only this many bytes and decode them. Subsequent calls to this function will provide the rest of the message. At the end, the destination struct will have the decoded message, as if those chunks had been written to a continuous buffer and decoded with `pb_decode`.

**In what situation would the feature be useful?**
This would be useful in very memory constrained applications using a ring/cyclic buffer. In that scenario, data can wrap around to the beginning of the buffer. If that data happened to be a proto message, it could be decoded only if it was copied to an intermediary buffer, this time continuous in memory. When memory is very scarce, this extra buffer is very expensive. Another scenario where this could be useful, is if an application reads a very long message in chunks (i.e. a socket that reads X bytes at a time). Instead of buffering all the data until it is complete, it could be immediately given to the decoder, again, saving precious memory.

PetteriAimonen, 2021-06-01:
In a sense, what you propose is already supported by the `pb_istream_t` callbacks. The callback can copy bytes to provided buffer from e.g. ring buffer, or wait for more bytes to come.

But there is a big architectural distinction: nanopb currently operates in "pull" mode: nanopb calls the callback and remains in control of the decoding process. What you propose would be "push" mode, where nanopb would take given amount of bytes and return back to user application.

I do agree that push mode would have several advantages, but it is also more difficult to implement - especially without breaking compatibility for decoding callbacks.

Rubeer, 2022-04-22:
If you look at the memory footprint from a whole system perspective and not a library perspective this is more important than it may seem.
I have a memory constrained embedded product that supports multiple connections (TCP listen socket). With the pull architecture, I am basically forced to create a separate task for each connection. With a push architecture, everything could be combined into a single task, saving a lot of memory (e.g. multiplex using select()).
I don't mind putting in some effort to work around this. Let's suppose I buffer the data and try to decode. If the decoding fails due to insufficient data, I simply try again on that connection when more data is received. Or is there a catch?
Is there any other library that supports this cleanly?

PetteriAimonen, 2022-04-22:
If you are fine with buffering the whole message, you can just wait until the complete message is received and decode it then. After all, you have to have some kind of framing to know the length of the protobuf message.

If buffering a whole message is too much, but buffering a single top-level field is ok, you can decode such a chunk at a time. Thanks to the mergeability of protobuf messages, the result will be the same as if the whole message was decoded at once. To figure out the field boundaries, use `pb_decode_tag()` and `pb_skip_field()`.

Real push mode support could take e.g. part of a long submessage at a time. Currently there is no easy way to do that with nanopb.

Of the other embedded protobuf implementations listed [here](https://jpa.kapsi.fi/nanopb/), none have direct support for push mode.

Rubeer, 2022-04-28:
Good point, I can use the framing that is already there. Thanks for the ideas.

closed 2022-05-30 by PetteriAimonen