nanopb/nanopb #671: Compiler warning handling in pb_decode.c

peter-toft-greve, 2021-06-03:
Hi devs
I found with a Qualcomm Hexagon (SDK3.0) compiler several compile warnings in pb_decode.c

`
../externals/nanopb/external/pb_decode.c:398:56: warning: comparison of constant 255 with enum 'pb_wire_type_t' (range [0, 7]) has undefined behavior [-Wenum-compare]
            if (wire_type != PB_WT_VARINT && wire_type != PB_WT_PACKED)
                                             ~~~~~~~~~ ^  ~~~~~~~~~~~~
../externals/nanopb/external/pb_decode.c:398:56: note: place two sets of parentheses around the constant to silence this warning
            if (wire_type != PB_WT_VARINT && wire_type != PB_WT_PACKED)
                                                       ^  ~~~~~~~~~~~~
                                                          ((          ))
../externals/nanopb/external/pb_decode.c:406:56: warning: comparison of constant 255 with enum 'pb_wire_type_t' (range [0, 7]) has undefined behavior [-Wenum-compare]
            if (wire_type != PB_WT_VARINT && wire_type != PB_WT_PACKED)
                                             ~~~~~~~~~ ^  ~~~~~~~~~~~~
../externals/nanopb/external/pb_decode.c:406:56: note: place two sets of parentheses around the constant to silence this warning
            if (wire_type != PB_WT_VARINT && wire_type != PB_WT_PACKED)
                                                       ^  ~~~~~~~~~~~~
                                                          ((          ))
../externals/nanopb/external/pb_decode.c:412:55: warning: comparison of constant 255 with enum 'pb_wire_type_t' (range [0, 7]) has undefined behavior [-Wenum-compare]
            if (wire_type != PB_WT_32BIT && wire_type != PB_WT_PACKED)
                                            ~~~~~~~~~ ^  ~~~~~~~~~~~~
../externals/nanopb/external/pb_decode.c:412:55: note: place two sets of parentheses around the constant to silence this warning
            if (wire_type != PB_WT_32BIT && wire_type != PB_WT_PACKED)
                                                      ^  ~~~~~~~~~~~~
                                                         ((          ))
../externals/nanopb/external/pb_decode.c:418:55: warning: comparison of constant 255 with enum 'pb_wire_type_t' (range [0, 7]) has undefined behavior [-Wenum-compare]
            if (wire_type != PB_WT_64BIT && wire_type != PB_WT_PACKED)
                                            ~~~~~~~~~ ^  ~~~~~~~~~~~~
../externals/nanopb/external/pb_decode.c:418:55: note: place two sets of parentheses around the constant to silence this warning
            if (wire_type != PB_WT_64BIT && wire_type != PB_WT_PACKED)
                                                      ^  ~~~~~~~~~~~~
                                                         ((          ))
4 warnings generated.

`

One fix is https://github.com/nanopb/nanopb/compare/master@%7B1day%7D...peter-toft-greve:master - are you interested to take this in? If so take it :)

PetteriAimonen, 2021-06-03:
Hmm, might be better to actually fix the undefined behavior, by moving PB_WT_PACKED definition from `pb_decode.c` to `pb.h` where the enum is defined.

I committed such change now, can you test if it fixes the warning?

peter-toft-greve, 2021-06-03:
Super with https://github.com/nanopb/nanopb/commit/98a4911bab0c7273824b4165e0633228e1cdf66a - I have verified that the fix is super

closed 2021-06-03 by peter-toft-greve