nanopb/nanopb #672: Conditional Decoding of field based on another field within same structure

vignesh0025, 2021-06-03:
Hi ,

I am trying to decode caffe.proto based buffer and I want to decode only certain fields based on another field.

I have tried the following:

```
bool print_layers(pb_istream_t *stream, const pb_field_t *field, void **arg)
{
    if(field->tag == caffe_NetParameter_layer_tag)
    {
        uint8_t buffer[1024] = {0};

        {
            caffe_LayerParameter layers = caffe_LayerParameter_init_zero;
            layers.name.funcs.decode = print_string;
            layers.name.arg = &buffer;
            if(!pb_decode(stream, caffe_LayerParameter_fields, &layers))
            {
                printf("Decoding each layer: \n");
            }
        }

        if(0 == strcmp((char*)buffer, "conv1_p"))
        {
            printf("we got: %s\n", (char*)buffer);
            caffe_LayerParameter layers = caffe_LayerParameter_init_zero;
            layers.blobs.funcs.decode = print_blobs;
            if(!pb_decode(stream, caffe_LayerParameter_fields, &layers))
            {
                printf("Decoding each blob: \n");
            }
            else
            {
                printf("Decoding each blob failed \n");
            }
        }
        /* layers.blobs.funcs.decode = print_blobs; */
    }
    else printf("No its not Layers");
    /* We could read block-by-block to avoid the large buffer... */
    return true;
}
```

Here, I want to decode **caffe_LayerParameter::blob** only if  **caffe_LayerParameter::name** is **conv1_p**. I am able to successfully find the "conv1_p" in the name field and if I call pb_decode again within the same callback, it fails. 

LayerParameter looks like this.
```
typedef struct caffe_LayerParameter { 
    pb_callback_t name; /* the layer name */
    pb_callback_t blobs; 
   .
   .
}
```

Please suggest how to achieve this.

PetteriAimonen, 2021-06-03:
Problem is that the first `pb_decode()` consumes the entire stream, so there is nothing left for the second call.

Fields in a protobuf message can come in any order, so in the general case if you are reading from a remote stream, you can't expect to receive `name` before `blob`.

But if your stream is just from memory buffer, you can take a copy of it using `pb_istream_t copystream = *stream;` and use that to decode for the first time.

Another option is to use `pb_decode_tag()` and `pb_read()` to read the `name` field only manually. This would work with any stream, but only if the `name` field occurs first.

vignesh0025, 2021-06-03:
Thank you for the reply,
Its a memory buffer which comes from a file (weights of a pretrained goturn neural network are stored in it ).  I want to extract weights of each layer. This entire operation is performed in an Arm a9 standalone platform.

```
bool print_layers(pb_istream_t *stream, const pb_field_t *field, void **arg)
{
    if(field->tag == caffe_NetParameter_layer_tag)
    {
        uint8_t buffer[1024] = {0};

        {
        pb_istream_t copystream = *stream;
        caffe_LayerParameter layers = caffe_LayerParameter_init_zero;
        layers.name.funcs.decode = print_string;
        layers.name.arg = &buffer;
        if(!pb_decode(&copystream, caffe_LayerParameter_fields, &layers))
            printf("Decoding each layer: \n");
        }

        if(0 == strcmp((char*)buffer, "conv1_p"))
        {
            caffe_LayerParameter layers = caffe_LayerParameter_init_zero;
            printf("we got: %s\n", (char*)buffer);
            layers.blobs.funcs.decode = print_blobs;
            if(!pb_decode(stream, caffe_LayerParameter_fields, &layers))
                printf("Decoding each blob: \n");
            else
                printf("Decoding each blob failed \n");
        }
        /* layers.blobs.funcs.decode = print_blobs; */
    }
    else printf("No its not Layers");
    /* We could read block-by-block to avoid the large buffer... */
    return true;
}
```

This time second decode call results in segmentation fault. Is my approach correct ?

closed 2022-05-30 by PetteriAimonen