nanopb/nanopb #673: Removal of trailing whitespace?

peter-toft-greve, 2021-06-03:
Are you interested in removal of trailing whitespace?
I think I have this fixed on https://github.com/peter-toft-greve/nanopb/tree/whitespace - take if you like.
Note that if you have quite a few parallel branches and ongoing PRs you probably want to not clean this up.

PetteriAimonen, 2021-06-03:
It is not generally worth the trouble. It just gets reintroduced either by me or someone else. I edit with various editors with little time to spare and most of them won't strip it.

(And to be honest, I haven't fully understood what is the problem caused by trailing whitespace)

peter-toft-greve, 2021-06-03:
@PetteriAimonen I fully understand - I suggest to close this now

closed 2021-06-03 by peter-toft-greve