nanopb/nanopb #675: Message framing with dynamic length messages

polfeliu, 2021-06-05:
Hello, 

I'm confused on how message framing works. From a fixed size message it seems straight forward, but how do you know the length of variable-length messages in advance? If I understood correctly callbacks, you can place anything you want inside it, the length may even be determined by an external IC. How does nanopb handle that?

https://jpa.kapsi.fi/nanopb/docs/concepts.html#message-framing

My respects to everyone that developed nanopb, I've been tinkering with it for a couple of weeks and it's awesome!

Pol

PetteriAimonen, 2021-06-05:
In many cases, encoding callbacks get called multiple times to first calculate the size and then actually generate the message. On the sizing calls, you can also just signal the length of the data without actually writing it (e.g. https://github.com/nanopb/nanopb/blob/master/pb_encode.c#L730 ).

But there are other ways also, for example you can use the zero-terminated format that is supported by nanopb (but not many other protobuf libraries), or if you are encoding to a memory buffer, you can prefix the length after the encoding.

polfeliu, 2021-06-05:
Oh, so I would need to have the data on a buffer to start transmitting it. That is inconvenient for me, I think I will try to implement a zero-terminated in the reception library. 
How does this work? it just adds a zero byte to the encoded sequence? How is it differenced from the rest of the frame?
My other end-point will be in python, with the official protobuf lib. If I develop it, do you think it's interesting to pull it into this repo to be reused by others?

PetteriAimonen, 2021-06-05:
Yes, zero termination works by appending a zero. It is detected by 0 not being a valid protobuf tag, even though it is allowed to occur at other positions of the message. I would be happy to see this being more widely supported, but I don't think it should be in nanopb repository. It will be pretty difficult to keep up to date with python-protobuf updates, and it is not something I would want to maintain. It would be great if such support could be merged to python-protobuf upstream.

Note though that such zero termination is not the only framing format that can avoid having a full memory buffer. You can also use e.g. chunked format, where you have a memory buffer of e.g. 256 bytes, fill it up and prefix with header telling the chunk size. End of message can be indicated by shorter chunk or zero length chunk, like in USB. Or you can use HDLC-style escaping format so that the message end marker cannot occur in message data.

closed 2022-05-30 by PetteriAimonen