nanopb/nanopb #680: static assertion failed: "FIELDINFO_DOES_NOT_FIT_width4_field1"

cc32d9, 2021-06-16:
bug1.proto:
```
syntax = "proto3";
message Transaction {
    bytes body = 1;
    repeated Signature signatures = 2;
}
enum KeyType {
    KEYTYPE0 = 0;
}
message Signature {
    KeyType type = 1;
    repeated bytes sig_bytes = 2;
}
```
bug1.options:
```
Transaction.body                       max_size:65536
Transaction.signatures                 max_count:8
Signature.sig_bytes                    max_size:128
Signature.sig_bytes                    max_count:16
```

this produces a compiler error:
```
In file included from bug1.pb.h:6,
                 from bug1.pb.c:4:
../pbtx/nanopb/pb.h:153:42: error: static assertion failed: "FIELDINFO_DOES_NOT_FIT_width4_field1"
  153 | #      define PB_STATIC_ASSERT(COND,MSG) _Static_assert(COND,#MSG);
      |                                          ^~~~~~~~~~~~~~
../pbtx/nanopb/pb.h:755:5: note: in expansion of macro ‘PB_STATIC_ASSERT’
  755 |     PB_STATIC_ASSERT(PB_FITS(tag,16) && PB_FITS(data_offset,16) && PB_FITS((int_least8_t)size_offset,8) && PB_FITS(data_size,16) && PB_FITS(array_size,16), FIELDINFO_DOES_NOT_FIT_width4_field ## tag)
      |     ^~~~~~~~~~~~~~~~
../pbtx/nanopb/pb.h:554:5: note: in expansion of macro ‘PB_FIELDINFO_ASSERT_4’
  554 |     PB_FIELDINFO_ASSERT_4(tag, PB_ATYPE_ ## atype | PB_HTYPE_ ## htype | PB_LTYPE_MAP_ ## ltype, \
      |     ^~~~~~~~~~~~~~~~~~~~~
bug1.pb.h:57:1: note: in expansion of macro ‘PB_GEN_FIELD_INFO_ASSERT_4’
   57 | X(a, STATIC,   SINGULAR, BYTES,    body,              1) \
      | ^
../pbtx/nanopb/pb.h:485:5: note: in expansion of macro ‘Transaction_FIELDLIST’
  485 |     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
      |     ^~~~~~~
bug1.pb.c:9:1: note: in expansion of macro ‘PB_BIND’
    9 | PB_BIND(Transaction, Transaction, 4)
      | ^~~~~~~
../pbtx/nanopb/pb.h:153:42: error: static assertion failed: "FIELDINFO_DOES_NOT_FIT_width4_field2"
  153 | #      define PB_STATIC_ASSERT(COND,MSG) _Static_assert(COND,#MSG);
      |                                          ^~~~~~~~~~~~~~
../pbtx/nanopb/pb.h:755:5: note: in expansion of macro ‘PB_STATIC_ASSERT’
  755 |     PB_STATIC_ASSERT(PB_FITS(tag,16) && PB_FITS(data_offset,16) && PB_FITS((int_least8_t)size_offset,8) && PB_FITS(data_size,16) && PB_FITS(array_size,16), FIELDINFO_DOES_NOT_FIT_width4_field ## tag)
      |     ^~~~~~~~~~~~~~~~
../pbtx/nanopb/pb.h:554:5: note: in expansion of macro ‘PB_FIELDINFO_ASSERT_4’
  554 |     PB_FIELDINFO_ASSERT_4(tag, PB_ATYPE_ ## atype | PB_HTYPE_ ## htype | PB_LTYPE_MAP_ ## ltype, \
      |     ^~~~~~~~~~~~~~~~~~~~~
bug1.pb.h:58:1: note: in expansion of macro ‘PB_GEN_FIELD_INFO_ASSERT_4’
   58 | X(a, STATIC,   REPEATED, MESSAGE,  signatures,        2)
      | ^
../pbtx/nanopb/pb.h:485:5: note: in expansion of macro ‘Transaction_FIELDLIST’
  485 |     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
      |     ^~~~~~~
bug1.pb.c:9:1: note: in expansion of macro ‘PB_BIND’
    9 | PB_BIND(Transaction, Transaction, 4)
      | ^~~~~~~
```

It works after changing to
```
Transaction.body                       max_size:65000
```

cc32d9, 2021-06-16:
the maximum value that still works: 65528

PetteriAimonen, 2021-06-16:
Try defining `PB_FIELD_32BIT` in either compiler settings or in pb.h.

cc32d9, 2021-06-16:
thanks, adding `-DPB_FIELD_32BIT` to the compiler helped. Still it will be great to have it recognize the situation automatically :)

PetteriAimonen, 2021-06-17:
I think automatic detection is not possible, because it would have to affect all files and not just the single `.pb.c` that has a huge message size. But a better error message should be added.

ShaharHD, 2022-02-28:
I'm having the exact same issue, even with simple protobuf protocol:

```protobuf
// [START declaration]

syntax = "proto3";
package HyGearBLEProtocol;

// [END declaration]

// [START messages]

// Calibration data maps a raw load cell value to a weight value in grams
message Calibration {
  reserved 1, 2;

  repeated sint32 loadcell_array = 3; // Array of raw loadcell values
  repeated sint32 weight_array = 4; // Arrays of actual weight in grams
}

// [END messages]
```

options file:

```protobuf
// * descriptorsize: DS_AUTO

// Max number of calibration points
Calibration.*_array max_count:10
Calibration.weight_array int_size:IS_16
```

generated code looks fine

```c
typedef struct _Calibration {
    pb_size_t loadcell_array_count;
    int32_t   loadcell_array[10]; /* Array of raw loadcell values */
    pb_size_t weight_array_count;
    int16_t   weight_array[10]; /* Arrays of actual weight in grams */
} HyGearBLEProtocol_Calibration;
```

Tried with both changing `* descriptorsize: DS_AUTO` as suggested in #757 and also using `PB_FIELD_32BIT` as suggested in #680

Using latest nanopb (v0.4.5) on MacOS M1 Max running latest monterey (v12.2.1)

PetteriAimonen, 2022-02-28:
@ShaharHD That message is not exceeding even 256 bytes, so should fit fine in any descriptor type.

What is the actual error message you are receiving?

ShaharHD, 2022-02-28:
> @ShaharHD That message is not exceeding even 256 bytes, so should fit fine in any descriptor type.
> 
> What is the actual error message you are receiving?

Warnings like this one:
`variably modified 'pb_static_assertion_FIELDINFO_DOES_NOT_FIT_width1_field2_9_2' at file scope`

I can re-run my example I did before if you want the actual warnings, but I can upload here a simple example project which you can run with Segger Embedded Studio on any platform.

Shahar

PetteriAimonen, 2022-02-28:
I won't be installing Segger Embedded Studio for free support, I don't have that much time available.

Sounds like the compiler used does not support the static asserts used.
If the settings have an option for enabling C11 standard, try turning it on.
Otherwise you can define `PB_NO_STATIC_ASSERT` to disable the checks.

ShaharHD, 2022-02-28:
> C11 standard,

`C11` resolved the issue - thanks!

Any idea why it just became an issue?

PetteriAimonen, 2022-02-28:
Some compilers use outdated standards by default.
`pb.h` uses classic negative size array hack for static asserts in that case, but some compilers are not smart enough to resolve it in every case.

closed 2022-05-30 by PetteriAimonen