nanopb/nanopb #682: including <pb.h> not compatible with bazel build

blumamir, 2021-06-25:
Hi,

I have a project using nanopb with bazel build system.
After compiling my `.proto` files to `.pb.h` the generated output has `include <pb.h>` in it which fails to compile with bazel, since it uses the following flags for gcc: `-iquotes external/nanopb` which expects the includes to have quotes format, e.g. `include "pb.h"`.

According to [this thread](https://github.com/RobotLocomotion/drake/issues/8197), to support including library files with system format (`-isystem`), the `BUILD.bazel` file should use `includes=[...]` instead of `hdrs=[...]` which is how the current file is written.

A workaround that worked for me is to specifically add `copts = ["-Iexternal/nanopb"]` to my `BUILD.bazel` file which has deps on `"@nanopb//:nanopb"`, so that gcc command can find the headers in `<pb.h>` format as well.

I am not a bazel expert, so might be missing something. Is there another way to fix the issue without extra configuration?

PetteriAimonen, 2021-06-25:
I know nothing about bazel, but there is a generator option `--library-include-format`.

@simmonmt Do you want to take a look?

blumamir, 2021-06-25:
Thanks @PetteriAimonen, 

Can you please point me to the documentation on how to use this generator option?

PetteriAimonen, 2022-05-30:
I don't know if there is something to fix in bazel rules here, but I'm closing this off for now.

closed 2022-05-30 by PetteriAimonen