nanopb/nanopb #683: package encode wrong data on arduino esp8266

losingrose, 2021-06-26:
```
syntax = "proto2";
message ModuleSwitch {
	required bool state = 1;
	optional fixed64 cycle_ms = 2;
	optional fixed64 alarm_poweroff = 3;
}
```

```
        ModuleSwitch packet = ModuleSwitch_init_zero;
        packet.state = false;
        packet.has_cycle_ms = true;
        packet.cycle_ms = cycle_ms;
        packet.alarm_poweroff = true;
        packet.alarm_poweroff = alarm_poweroff;
        pb_ostream_t stream = pb_ostream_from_buffer(msg_buf, sizeof(msg_buf));
        pb_encode(&stream, ModuleSwitch_fields, &packet);
        client->publish(topic.c_str(), msg_buf, stream.bytes_written);
        Serial.println(stream.bytes_written);
```
        
stream.bytes_written always is 3;
server received data buffer is : [ 8, 0, 17]

PetteriAimonen, 2021-06-26:
Maybe `msg_buf` is too small?

losingrose, 2021-06-26:
msg_buf is global buffer，another Log struct with a string field work correct。
```
message Log {
	required fixed64 time = 1;
	required string msg = 2;
}
```

PetteriAimonen, 2021-06-26:
If `pb_encode()` is returning false, check the error message in `stream.errmsg`.

losingrose, 2021-06-26:
Thanks, I found the problem. "stream full"
I pass the msg_buf point to this class，the sizeof(msg_buf) is wrong value 4.

closed 2021-06-26 by losingrose