nanopb/nanopb #684: Enclosing tags in enums for oneof which_ type safety

polfeliu, 2021-07-04:
Hello everyone,

Is there any reason tags are declared in defines rather that enums?

tags could be grouped in enum types, that more explicitly indicate what message they are for

```proto
message message1{
    required hello = 1;
    oneof lorem{
        foo = 3;
        bar = 6
    }
}
```

```c
typedef enum{
    message1_hello_tag = 1
} message1_global_tags

typedef enum{
    message1_foo_tag = 3,
    message1_bar_tag = 6
} message1_oneof_lorem_tags;
```
Conserving the naming of the enum fields would not break programs developed with nanopb. 
This may require to introduce a cast (pb_size_t) here and there in nanopb (?)

This has the ENORMOUS advantage of then being able to type check the which field on protobuf oneofs, by changing the type of "pb_size_t" to message1_oneof_lorem_tags;

If the type is not correct, compilation fails :).
With defines you can put whatever define your please and produce unexpected results. This is much safer from a developer stand of view

PetteriAimonen, 2021-07-04:
I agree with the reasoning.

The reason why it is the way it is, is that the tag number `#define` were first, and oneof support came later.

Note however that the tag defines are generated for all fields, and can be used in e.g. encoding callbacks and other user code. For type safety, each enum for a oneof should only contain the valid values for that field, plus a `0` value for "unset". But then it seems weird to have `#define` for some fields and `enum` for other fields.

polfeliu, 2021-07-04:
I'm not that familiar with encoding callbacks. I've only used some of the examples.
Is it something that it would be interesting to type check? 

I think there are multiple approaches to group types. Maybe having a global enum that doesn't contain all the fields is a bit weird. What if it would contain all of them, and then oneof contains a subset of these, pointing to the global enum?

If oneof is the only interesting thing to type globals could be left as defines (not ideal in my opinion)

PetteriAimonen, 2021-07-04:
The main use (apart from oneofs) for the defines is passing to functions like `pb_encode_tag()`. There probably isn't a reasonable way to type check these because it is not tied to any particular message structure at that point. Many compilers will also warn if an enum is passed as a parameter to function expecting `uint32_t`. In reality this use is probably pretty rare, but it's used often enough that compatibility needs to be preserved.

But if most fields are left as global defines, then either the enum-based oneof tags would have to have different names, or the defines for those would have to be omitted. Either way feels weird.

polfeliu, 2021-07-05:
I don't see different naming for oneofs as a problem. Having the name of the oneof in the tag name is nice. Maybe we could also omitt the "tag" suffix. 

I agree that it doesn't make much sense to type pb_encode_tag. This then would be something only related to oneofs.

If I manage to develop it would you review a PR? I'd prefer not to develop with a forked library

PetteriAimonen, 2021-07-05:
PRs are always welcome, though I must warn that the bar of acceptance is quite high for this. I consider it as a quite low priority enhancement, because it does not make possible anything new. It merely avoids a particular type of bug, and the bug itself does not have any severe consequences.

My gut feeling is that the feature will be quite difficult to implement in a way that does not cause compatibility issues. Since yesterday, two new concerns have come to my mind:

1. How will the pb_encode / pb_decode code know the size of the enum field? C compiler is free to select how many bytes it will use for enum, while `pb_size_t` is always constant width. It would be possible to add a fake `PB_SIZE_MAX` entry to all of the enum definitions, but that seems ugly and does not prevent the compiler from using a larger 32 or 64-bit type.
2. If enums will have different names than current `#define`s do, will it cause warnings when existing code is compiled? Causing a bunch of false positive type safety warnings will just lead to a bunch of useless work and will make people ignore them.

polfeliu, 2021-07-05:
I do think it has consequences, but it doesn't avoid them entirely, just one type. Oneofs with which field and union can be dangerous if not properly synced. What if you the decoder calls a callback from another union member, which is not the type that has been set? Program jumps to a random address and crashes.
I understand your concerns and I haven't found an elegant answer either. I'm now leaning more towards assigning oneofs with a processor directive that somehow would check this. 100% speculation, not sure how to implement it