nanopb/nanopb #687: FIELDINFO_DOES_NOT_FIT_width4_field2 error even though fields should fit into 16 bit

fjhunter, 2021-07-09:
Hi, 
I don't understand why i get the error `static assertion failed: "FIELDINFO_DOES_NOT_FIT_width4_field2"`, even tough every field should fit in 16 bit.

The DataSet message causes the problem in the following `.proto` file:

```
syntax = "proto2";
package test;
message DataPoint {
  required uint64 timestamp = 1;
  required int32 value = 2;
}

message DataSet {
  repeated DataPoint data = 1;
  required bool more = 2;
}
```

And the option file:

```
test.DataSet.data    max_count:16384
```

It generates the following code:

```
typedef struct _test_DataSet { 
    pb_size_t data_count;
    test_DataPoint data[16384]; 
    bool more; 
} test_DataSet;

typedef struct _test_DataPoint { 
    uint64_t timestamp; 
    int32_t value; 
} test_DataPoint;
```

The generated code seems right. The max value `data_count` can have is `16384` and `16384` should fit into 16 bit. Am I missing something? Why do i get this error?

PetteriAimonen, 2021-07-09:
While 16384 fits into 16 bits, the size of the structure will be `16384 * sizeof(test_DataPoint) = 16384 * 12 = 196 608`, which doesn't fit.

Define `PB_FIELD_32BIT`, like in issue #680.

fjhunter, 2021-07-09:
Ah okay. Now I get it. I was just confused because I thought only the `max_count` has to fit into 16 bits. Thanks a lot @PetteriAimonen

closed 2021-07-09 by fjhunter

PetteriAimonen, 2021-07-09:
Yeah, I think the error comes from the `more` field. Its offset from the start of the struct will be too large to fit in 16 bits.

fjhunter, 2021-07-09:
This makes sense. Thanks