nanopb/nanopb #688: How to solve the problem of No module named google.protobuf.text_format

c20liu, 2021-07-20:
Hi, All.
 
When I used "--plugin=protoc-gen-nanopb=......./generator/protoc-gen-nanopb-py2"  I encountered the error "No module named google.protobuf.text_format", I tried many ways but failed to. solve it,  who can help and guide me to solve it?

My environment:

[centos@k8s-master simple]$ protoc --version
libprotoc 3.17.3

[centos@k8s-master simple]$ python --version
Python 3.9.6

[centos@k8s-master simple]$ pip --version
pip 21.1.3 from /usr/local/python3/lib/python3.9/site-packages/pip (python 3.9)

[centos@k8s-master simple]$ pip install protobuf==3.17.3
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: protobuf==3.17.3 in /usr/local/python3/lib/python3.9/site-packages (3.17.3)
Requirement already satisfied: six>=1.9 in /usr/local/python3/lib/python3.9/site-packages (from protobuf==3.17.3) (1.16.0)

I doing the t**wo-line command to build .proto files**.  and it is successful
protoc -osimple.pb simple.proto
python ../../generator/nanopb_generator.py simple.pb

but when I do **a one-line command to build .proto files**.  it is failed and report:

[centos@k8s-master simple]$ protoc --plugin=protoc-gen-nanopb=/home/centos/gopath/src/github.com/nanopb/nanopb/generator/protoc-gen-nanopb-py2  --nanopb_out=.  simple.proto

         *************************************************************
         *** Could not import the Google protobuf Python libraries ***
         *** Try installing package 'python3-protobuf' or similar.  ***
         *************************************************************

Traceback (most recent call last):
  File "/home/centos/gopath/src/github.com/nanopb/nanopb/generator/nanopb_generator.py2", line 6, in <module>
    from nanopb_generator import *
  File "/home/centos/gopath/src/github.com/nanopb/nanopb/generator/nanopb_generator.py", line 36, in <module>
    import google.protobuf.text_format as text_format
ImportError: No module named google.protobuf.text_format
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.

I can find "google.protobuf.text_format" in /usr/local/python3/lib/python3.9/site-packages

PetteriAimonen, 2021-07-20:
On this line:

    protoc --plugin=protoc-gen-nanopb=/home/centos/gopath/src/github.com/nanopb/nanopb/generator/protoc-gen-nanopb-py2 --nanopb_out=. simple.proto

The `protoc-gen-nanopb-py2` is explicitly requesting the Python 2 version of the generator. Try replacing that with `protoc-gen-nanopb`.

c20liu, 2021-07-21:
@PetteriAimonen  thanks for your help.  it is solved my problem.  
(Jump over from other projects,  not very familiar with python and this project)

closed 2021-07-21 by c20liu