nanopb/nanopb #690: NANOPB_VERSION as string

golgoth31, 2021-07-23:
Hi, I tried to use the define "NANOPB_VERSION" value but I have an error at compile time. Converting it as string is resolving this issue:
before: #define NANOPB_VERSION nanopb-x.y.z
after: #define NANOPB_VERSION "nanopb-x.y.z"

I make a mistake using this value ?

PetteriAimonen, 2021-07-23:
Yeah, it might make more sense as a string directly.

One can convert to string with preprocessor `#` though.

golgoth31, 2021-07-23:
I'am quite new C so i just "discover" macro expansion with # :)

but, having directly it to string should be little bit simpler :)
Do you want a PR for this ?

PetteriAimonen, 2021-07-23:
Yeah, I'm just wondering whether it has any backwards compatibility concerns. Probably not.

A brief search on [sourcegraph](https://sourcegraph.com/search?q=context:global+NANOPB_VERSION&patternType=literal&case=yes) seems to indicate that it is not even used much, and adding the string delimiters would just add extra `"` in the strings.

I guess I'd be ok with a PR that has:

1. The change for `tools/set_version.sh`
2. Added note in `docs/migration.md`

PR #691 (2021-07-24): feat: Change NANOPB_VERSION macro to string
closes #690
closed 2021-07-24 by PetteriAimonen