nanopb/nanopb #692: Duplicate declarations of size_unions with repeated fields inside a oneof

IngoKresse, 2021-07-26:
This seems to be another incarnation of #637, tested against version 0.4.5 and master. Most of the investigations and the guards were done by my colleague Henrik Poppe (@AlmostAll), thanks a lot!

If one part of a oneof is a message in another file which has a repeated field (with a max_count option), then the declaration of the size_union may be emitted multiple times. Here are example messages that reproduce the problem:

a.proto
```protobuf
syntax = "proto3";

import "b.proto";

package a;

message A {
  oneof data {
     int32 number = 1;
     b.BigList1 message = 2;      // unknown max_count at this point (?) -> generates size_union
  }
}

message AA {
  b.BigList2 another_message = 2; // second sub-message of unknown size -> causes second guard block
  A sub_message = 1;              // here, the size_union gets duplicated :-(
}
```

b.proto
```protobuf
syntax = "proto3";

package b;

message BigList1 {
  repeated double numbers = 1;
}

message BigList2 {
  repeated double more_numbers = 1;
}
```

b.options
```
b.BigList1.numbers max_count: 5;
b.BigList2.more_numbers max_count: 6;
```

**What happens?**

The following code is generated inside a.pb.h:

```cpp
/* Maximum encoded size of messages (where known) */
#if defined(b_BigList1_size)
#define a_A_size                                 (0 + sizeof(union a_A_data_size_union))
union a_A_data_size_union {char f2[(6 + b_BigList1_size)]; char f0[11];};
#endif
#if defined(b_BigList1_size) && defined(b_BigList2_size)
#define a_AA_size                                (12 + sizeof(union a_A_data_size_union) + b_BigList2_size)
union a_A_data_size_union {char f2[(6 + b_BigList1_size)]; char f0[11];};
#endif
```

The union `a_A_data_size_union` is declared twice and the compiler throws an error.

**What should happen?**

I see several possibilities  how this could be solved:

1. If the generator could read the options for the dependencies, then it may be able to compute the maximum sizes directly.
2. The generator could only emit the size_union declaration together with the message where the oneof is defined, in a patch like this:
```diff
--- ./old/nanopb_generator.py<->2021-03-22 14:44:15.000000000 +0000
+++ ./new/nanopb_generator.py<->2021-07-23 13:14:12.988435713 +0000
@@ -1811,7 +1811,13 @@
                     if cpp_guard not in guards:
                         guards[cpp_guard] = set()
                     for decl in msize.get_declarations().splitlines():
-                        guards[cpp_guard].add(decl)
+                        if decl.startswith('union'):
+                            size_union_msg_name = '_'.join(decl.split(' ')[1].split('_')[:-3])
+                            # emit size_unions only once - with the size declarations where it belongs
+                            if size_union_msg_name == identifier[:-5]:
+                                guards[cpp_guard].add(decl)
+                        else:
+                            guards[cpp_guard].add(decl)
                     guards[cpp_guard].add('#define %-40s %s' % (identifier, msize))
                 else:
                     yield '/* %s depends on runtime parameters */\n' % identifier

```
but this declaration could be in another header file. What happens, if that declaration is not included before? This would lead to compile errors. I am not sure if all corner cases would be covered.

3. A very simplistic solution is to add guards around the size_union declarations, to avoid multiple declarations:
```diff
--- ./old/nanopb_generator.py<->2021-03-22 14:44:15.000000000 +0000
+++ ./new/nanopb_generator.py<->2021-07-23 13:21:13.440435713 +0000
@@ -1811,7 +1811,12 @@
                     if cpp_guard not in guards:
                         guards[cpp_guard] = set()
                     for decl in msize.get_declarations().splitlines():
-                        guards[cpp_guard].add(decl)
+                        if decl.startswith('union'):
+                            # emit guards for union declarations
+                            size_union_guard = decl.split(' ')[1].upper()
+                            guards[cpp_guard].add('#ifndef ' + size_union_guard + '\n#define ' + size_union_guard + '\n' + decl + '\n#endif')
+                        else:
+                            guards[cpp_guard].add(decl)
                     guards[cpp_guard].add('#define %-40s %s' % (identifier, msize))
                 else:
                     yield '/* %s depends on runtime parameters */\n' % identifier
```

which leads to code like this:

```cpp
/* Maximum encoded size of messages (where known) */
#if defined(b_BigList1_size)
#define a_A_size                                 (0 + sizeof(union a_A_data_size_union))
#ifndef A_A_DATA_SIZE_UNION
#define A_A_DATA_SIZE_UNION
union a_A_data_size_union {char f2[(6 + b_BigList1_size)]; char f0[11];};
#endif
#endif
#if defined(b_BigList1_size) && defined(b_BigList2_size)
#define a_AA_size                                (12 + sizeof(union a_A_data_size_union) + b_BigList2_size)    /* could this be (12 + a_A_size + b_BigList2_size) ? */
#ifndef A_A_DATA_SIZE_UNION
#define A_A_DATA_SIZE_UNION
union a_A_data_size_union {char f2[(6 + b_BigList1_size)]; char f0[11];};
#endif
#endif
```
This may look a bit messy, but is certainly safe.

What do you think?

PetteriAimonen, 2021-07-26:
Maybe the union should be emitted inside its own guard that include all of its immediate dependencies, and nothing else.

So in this case, it would be inside `#if defined(b_BigList1_size)`.
And in the general case of oneof including multiple unknown message sizes, it would have a guard that requires all of those, but nothing else. That should guarantee that the definition is available whenever possible.

IngoKresse, 2021-07-26:
I must admit that I am a bit confused - I saw in version 0.4.2, that the size_union construct appeared in a different file from where the message with a oneof was declared (causing a duplicate-declaration-compile-error). This does not happen to me with 0.4.5 anymore.

From version 0.4.5 this appears only to be a problem in the same file? Then the pull request #693 above could indeed fix the problem...

PetteriAimonen, 2021-07-26:
0.4.5 had a fix for #610 and #637 which changed how the unions are generated.

IngoKresse, 2021-07-27:
Thanks a lot for your quick replies!

closed 2022-05-30 by PetteriAimonen