nanopb/nanopb #694: Fail to compile test code

passerbyd, 2021-08-06:
I'm trying to run "alltypes_proto3" test case seperately:
1. copy alltypes.proto and encode_alltypes.c into examples/simple.
2. modify Makefile, point source and proto files to encode_alltypes.c and alltypes.proto.
3. run make.

Then there are numbers of errors like:
> encode_alltypes.c:17:14: error: no member named 'rep_int32_count' in 'struct _AllTypes'
>     alltypes.rep_int32_count = 5; alltypes.rep_int32[4] = -2001;
>     ~~~~~~~~ ^
> encode_alltypes.c:17:53: error: subscripted value is not an array, pointer, or vector
>     alltypes.rep_int32_count = 5; alltypes.rep_int32[4] = -2001;
>                                   ~~~~~~~~~~~~~~~~~~^~
> encode_alltypes.c:18:14: error: no member named 'rep_int64_count' in 'struct _AllTypes'
>     alltypes.rep_int64_count = 5; alltypes.rep_int64[4] = -2002;
>     ~~~~~~~~ ^
> encode_alltypes.c:18:53: error: subscripted value is not an array, pointer, or vector
>     alltypes.rep_int64_count = 5; alltypes.rep_int64[4] = -2002;
>                                   ~~~~~~~~~~~~~~~~~~^~

and I checked the generated alltypes.pb.h file. All "repeated" and "string" fields are declared as "pb_callback_t", which can not be assigned with those values.

Please help me to figure out which step is wrong.

PetteriAimonen, 2021-08-06:
Did you copy `alltypes.options` also?

passerbyd, 2021-08-07:
> 
> 
> Did you copy `alltypes.options` also?

Didn't. I will have a try. But does the generated message struct will be different with options file. The problem is caused because the test code wants to assign an int value to "pb_callback_t" object.

PetteriAimonen, 2021-08-07:
Yeah, that is exactly what the options do: https://jpa.kapsi.fi/nanopb/docs/concepts.html#data-types

passerbyd, 2021-08-09:
All works fine since adding options file. Thx.

closed 2021-08-09 by passerbyd