nanopb/nanopb #697: Signalling EOF for istream with custom callback

blumamir, 2021-08-14:
Hello,
Thanks for the great library,

We use nanopb on ESP32 and read a proto stream from file into nanopb input stream,
we wrote an istream similar to the one in the [documentation](https://jpa.kapsi.fi/nanopb/docs/concepts.html).

According to the documentation: 
> You don’t need to know the length of the message in advance. After getting EOF error when reading, set bytes_left to 0 and return false. pb_decode() will detect this and if the EOF was in a proper position, it will return true.

but the example in the documentation does not follow the above rule as it returns `status`, which is expected to be `true`:
```cpp
bool callback(pb_istream_t *stream, uint8_t *buf, size_t count)
{
   FILE *file = (FILE*)stream->state;
   bool status;

   if (buf == NULL)
   {
       while (count-- && fgetc(file) != EOF);
       return count == 0;
   }

   status = (fread(buf, 1, count, file) == count);

   if (feof(file))
       stream->bytes_left = 0;

   return status;
}

pb_istream_t stdinstream = {&callback, stdin, SIZE_MAX};
```

In reality, both ways are incorrect as nanopb keeps calling the callback after setting `bytes_left` to 0.

We saw in [the code](https://github.com/nanopb/nanopb/blob/master/pb_decode.c#L116) that after callback is invoked `bytes_left` is decreased by `count` which we suspect causes it to be != 0,
setting `bytes_left` to `count` instead of 0 when we reach EOF in the callback solved the issue for us,

Is this a known issue? Are we using the callback incorrectly?
Any insight on the subject will be much appriciated.

PetteriAimonen, 2021-08-15:
Hmm yeah, looks like there is some ambiguity in the documentation specs and also in how the code works.

The inconsistency is when the last read is successful, but EOF is already known.

The following workarounds are ok:

1. Setting `bytes_left` to `count` when the read is successful, and setting it to `0` when read is unsuccessful because of EOF.
2. Setting `bytes_left` to `0` only when the read is unsuccessful because of EOF.

The proper fix will be to change `pb_read()` to check if `bytes_left` is below `count` before subtracting.

blumamir, 2021-08-15:
Thank you for the fast reply and for looking into this.
We can offer help with updating the documentation later this week so the example there will correctly reflect the described flows (basically, if I'm not mistaking, moving the `feof` check to happen before `fread`).

blumamir, 2021-08-15:
> 2\. Setting `bytes_left` to `0` only when the read is unsuccessful because of EOF.

Haven't tried it myself, but in this case, when returning `false` due to `EOF`, wouldn't the current implementation branch into the `if` and fail the decode due to `PB_RETURN_ERROR(stream, "io error")`?
I think that if callback returns `false`, `pb_read` should check the value in `bytes_left` and return `true` if it is 0 (while not modifying `bytes_left`).
Something like this:
```cpp
    if (!stream->callback(stream, buf, count)) {
        if(stream->bytes_left == 0) {
            return true;
        } else {
            PB_RETURN_ERROR(stream, "io error"); 
        }
   }
    stream->bytes_left -= count;
```

PetteriAimonen, 2021-08-15:
No, `pb_read()` is supposed to return false whenever it is unable to read `count` bytes.

If (and only if) this occurs at the point where a new field should begin, [pb_decode_varint32_eof](https://github.com/nanopb/nanopb/blob/master/pb_decode.c#L168) will detect it and end the decoding without error. If EOF occurs in the middle of a field, it is important that this is reported as error.

PetteriAimonen, 2021-08-15:
The impact of this bug is not very large in most cases, because usually even when it is triggered, this happens:

1. `pb_read()` is called to read the last `count` bytes successfully.
2. `bytes_left` is incorrect value at this point due to overflow.
3. `pb_read()` is called again to read next bytes, and e.g. `fread()` will still report an error and at this time EOF is correctly detected because `false` is returned.

I guess in your specific case, something happens in step 3. that causes the EOF condition not to be detected again.

blumamir, 2021-08-15:
Thank you, I need to dive deeper into the code to fully understand the flow.

> I guess in your specific case, something happens in step 3. that causes the EOF condition not to be detected again.

My callback is similar to the one in the documentation where I first execute `fread` and only then check for `EOF`, which then sets `bytes_left` to 0 and returns false which fails decoding.

Anyhow, I now have the understanding of how to workaround my specific issue and why it works.
Appreciate the help on this

blumamir, 2021-08-18:
Hello @PetteriAimonen 
We debugged this again and were able to workaround the issue to make it work.
However, after logging a bit, we are unsure what is the full extent of the issue we were facing.

Just in case it will help anyone, we paste the code and log here.
We won't pursue this any further right now, but feel free to reach out if any more information is needed or we can help in any other way.
```cpp
bool FileRead_callback(pb_istream_t *stream, uint8_t *buf, size_t count)
{
    File *file = (File*)stream->state;
    bool status;

    Serial.print("available: ");
    Serial.println(file->available());
    Serial.print("stream->bytes_left: ");
    Serial.println(stream->bytes_left);
    Serial.print("count: ");
    Serial.println(count);

    size_t bytesRead = file->read(buf, count);
    Serial.print("bytesRead: ");
    Serial.println(bytesRead);
    Serial.println("");

    status = (bytesRead == count);
    if (file->available() == 0)
        stream->bytes_left = 0;

   return status;
}

```

and the prints:
```
available: 36
stream->bytes_left: 4294967295
count: 1
bytesRead: 1

available: 35
stream->bytes_left: 4294967294
count: 4
bytesRead: 4

available: 31
stream->bytes_left: 4294967290
count: 1
bytesRead: 1

available: 30
stream->bytes_left: 4294967289
count: 1
bytesRead: 1

available: 29
stream->bytes_left: 4294967288
count: 1
bytesRead: 1

available: 28
stream->bytes_left: 4294967287
count: 1
bytesRead: 1

available: 27
stream->bytes_left: 12
count: 1
bytesRead: 1

available: 26
stream->bytes_left: 11
count: 1
bytesRead: 1

available: 25
stream->bytes_left: 10
count: 4
bytesRead: 4

available: 21
stream->bytes_left: 6
count: 1
bytesRead: 1

available: 20
stream->bytes_left: 5
count: 1
bytesRead: 1

available: 19
stream->bytes_left: 4
count: 1
bytesRead: 1

available: 18
stream->bytes_left: 3
count: 1
bytesRead: 1

available: 17
stream->bytes_left: 2
count: 1
bytesRead: 1

available: 16
stream->bytes_left: 1
count: 1
bytesRead: 1

available: 15
stream->bytes_left: 4294967274
count: 1
bytesRead: 1

available: 14
stream->bytes_left: 4294967273
count: 1
bytesRead: 1

available: 13
stream->bytes_left: 13
count: 1
bytesRead: 1

available: 12
stream->bytes_left: 12
count: 1
bytesRead: 1

available: 11
stream->bytes_left: 11
count: 5
bytesRead: 5

available: 6
stream->bytes_left: 6
count: 1
bytesRead: 1

available: 5
stream->bytes_left: 5
count: 1
bytesRead: 1

available: 4
stream->bytes_left: 4
count: 1
bytesRead: 1

available: 3
stream->bytes_left: 3
count: 1
bytesRead: 1

available: 2
stream->bytes_left: 2
count: 1
bytesRead: 1

available: 1
stream->bytes_left: 1
count: 1
bytesRead: 1

available: 0
stream->bytes_left: 4294967295
count: 16
bytesRead: 0

callback failed
```

PetteriAimonen, 2021-08-19:
It seems that your `file->read()` keeps successfully reading even after EOF has been reached, which is why you are hitting this bug.

Also, if you do know the file length in advance (e.g. `file->available()` is valid already when you open the stream), you can just set that to `bytes_left` when you open the stream and nanopb will obey it.

blumamir, 2021-08-19:
> It seems that your `file->read()` keeps successfully reading even after EOF has been reached, which is why you are hitting this bug.

```
available: 1
stream->bytes_left: 1
count: 1
bytesRead: 1

available: 0
stream->bytes_left: 4294967295
count: 16
bytesRead: 0
```

The last read is for `count: 16`, but `available: 0`, so `bytesRead: 0` as expected (was asked to read 16 bytes but read 0) 
So unless I'm missing something it fails the last read where no more bytes are available and returns 0 which casues the callback to return false (failure) with `stream->bytes_left: 0`

> Also, if you do know the file length in advance (e.g. `file->available()` is valid already when you open the stream), you can just set that to `bytes_left` when you open the stream and nanopb will obey it.

Yes thanks. That's what we ended up doing, but I'm still puzzled by why it did not work with the EOF pattern

PetteriAimonen, 2021-08-19:
Hmm yeah, something weird is going on with that log that does not exactly match my current understanding of the bug.

But good that you have it working now, I'll take a closer look at this some time!

gordonklaus, 2022-06-30:
I ran into what I think is this same issue.  The problem was that my `feof` function had (what I understand to be) ISO C rather than ANSI C semantics; that is, it returned true as soon as end of file was reached (whereas in ANSI C it only returns true after trying to read past the end).  Once I adjusted my `feof` implementation, the problem was fixed.

(I had originally synthesized my own `feof` because Zephyr's filesystem doesn't provide it.  The solution was just to interpret `fs_read(...) == 0` as EOF, which seems obvious is hindsight.)

I think nanopb should allow ISO C's `feof` semantics.  Alternatively, the documentation could be made clearer on this point.

FWIW, my read callback implementation:
```C
static bool read_callback(pb_istream_t *stream, uint8_t *buf, size_t count) {
	struct fs_file_t *f = stream->state;
	int ret = fs_read(f, buf, count);
	if (ret < 0) {
		LOG_ERR("fs_read: %d", ret);
		return false;
	}
	if (ret == 0) {
		stream->bytes_left = 0;
		return false;
	}
	return true;
}
```

closed 2022-12-11 by PetteriAimonen