nanopb/nanopb #698: CMake NANOPB_GENERATE_CPP_APPEND_PATH not working as intended

gautamr95, 2021-08-18:
I believe this is related to issue #527.

I'm trying to compile protos for an app that uses several nested proto packages using CMake. The directory tree looks something like:

```
proto
├── A
│   └── A.proto
├── B
│   └── B.proto
├── C.proto
└── pkg1
    └── subpkg1
        ├── D.proto
        ├── E.proto
        ├── subsubpkg1
        │   ├── F.proto

app
├── CMakeLists.txt
├── build
```

A, B, and C are independent proto files (no imports), but proto files within pkg1 may import each other.
For example, D.proto may contain:
```
import "pkg1/subpkg1/subsubpkg1/F.proto"
```

Now my CMakeLists.txt contains the following:
```
set(ROOT_PROTO_FILES_PATH ${CMAKE_CURRENT_SOURCE_DIR}/../proto)
FILE(GLOB SUB1_PROTO_FILES ${ROOT_PROTO_FILES_PATH}/pkg/subpkg1/*.proto)
FILE(GLOB SUBSUB1_PROTO_FILES ${ROOT_PROTO_FILES_PATH}/pkg/subpkg2/*.proto)

nanopb_generate_cpp(SUB1_PROTO_SRCS SUB1_PROTO_HDRS RELPATH ${ROOT_PROTO_FILES_PATH}/ ${SUB1_PROTO_FILES})
nanopb_generate_cpp(SUBSUB1_PROTO_SRCS SUBSUB1_PROTO_HDRS RELPATH ${ROOT_PROTO_FILES_PATH}/ ${SUBSUB1_PROTO_FILES})

target_include_directories(app PRIVATE ${CMAKE_CURRENT_BINARY_DIR})
target_sources(app PRIVATE ${SUB1_PROTO_SRCS})
target_sources(app PRIVATE ${SUBSUB1_PROTO_SRCS})
```

This generally builds fine and places all the `pb.c` and `pb.h` files within the `build/` directory directly (not nested based on real directory structure). This is fine for A, B, and C as there are no dependencies. However, the variables `SUB1_PROTO_SRCS` and `SUB1_PROTO_HDRS` (and subsub1 equivalents) seem to expect the same directory structure as whats in the `proto` folder.

So I get errors that look like:
```
build/pkg/subpkg1/D.pb.c: No such file or directory
build/pkg/subpkg1/subsubpkg1/F.pb.c: No such file or directory
```
Because all compiled proto files are placed directly in `build/`

So based on issue #527 I tried adding this line:
```
set(NANOPB_GENERATE_CPP_APPEND_PATH FALSE)
```
but this breaks compilation of A, B, and C:
```
proto/A/A.proto: File does not reside within any path specified using --proto_path (or -I).  You must specify a --proto_path which encompasses this file.  Note that the proto_path must be an exact prefix of the .proto file names -- protoc is too dumb to figure out when two paths (e.g. absolute and relative) are equivalent (it's harder than you think).
```

Now what's especially confusing to me is that the [cmake_relpath example](https://github.com/nanopb/nanopb/tree/master/examples/cmake_relpath) works exactly as intended, with the proto files generated within `build/` using the same directory structure as the `proto` folder. This example also does not set `NANOPB_GENERATE_CPP_APPEND_PATH` to `false`, which contradicts the comment here: https://github.com/nanopb/nanopb/blob/bdfaeaa09501f520953372e4d6043c9dad533c79/extra/FindNanopb.cmake#L22

Also when making the change suggested here: https://github.com/nanopb/nanopb/blob/bdfaeaa09501f520953372e4d6043c9dad533c79/extra/FindNanopb.cmake#L255
everything compiles as expected, within the right subfolders in my project, but then the sample breaks.

This is my first time using CMake so I'm not too familiar with what's available but I don't understand the discrepancy :confused:.

Versions:
`cmake`: 3.21.1
`protoc`: 3.6.1
`ubuntu`: 20.0.4

gautamr95, 2021-08-18:
Hmm perhaps it's related to this line:https://github.com/nanopb/nanopb/blob/bdfaeaa09501f520953372e4d6043c9dad533c79/extra/FindNanopb.cmake#L136

Since my protos are not located in the same directory as my app, they do not get correctly included via `-I`

gautamr95, 2021-08-18:
Phew I think I finally figured this out:

Turns out protoc does not like relative paths (with ../).

So I changed:
```
set(ROOT_PROTO_FILES_PATH ${CMAKE_CURRENT_SOURCE_DIR}/../proto)
```
to
```
get_filename_component(ROOT_PROTO_FILES_PATH ${CMAKE_CURRENT_SOURCE_DIR}/../../../common/proto ABSOLUTE)
```
and also set:
```
set(NANOPB_GENERATE_CPP_APPEND_PATH FALSE)
```
And the pb.c and pb.h files built in the right directories! Still not sure why the example did this without needing to set `NANOPB_GENERATE_CPP_APPEND_PATH` to `FALSE` but I'm at least unblocked!

Hopefully this helps anyone else with a similar issue.

closed 2021-08-18 by gautamr95

meuter, 2022-01-31:
Thanks @gautamr95 ! I was facing the exact same issue ;-)

BTW, in recent versions of cmake, you can use `file(GLOB_RECURSE ...)` to recursively find all files in a subdirectory. So in your case, you could simplify things quite a bit to something like:

```cmake
file(GLOB_RECURSE ALL_PROTO_FILES ${ROOT_PROTO_FILES_PATH}/*.proto)
nanopb_generate_cpp(ALL_PROTO_SOURCE ALL_PROTO_HEADERS RELPATH ${ROOT_PROTO_FILES_PATH}/ ${ALL_PROTO_FILES})
target_include_directories(app PRIVATE ${CMAKE_CURRENT_BINARY_DIR})
target_sources(app PRIVATE ${ALL_PROTO_SOURCE})
```

gautamr95, 2022-02-01:
Glad this was useful @meuter! Thanks for the suggestion I'll check it out for my project!