nanopb/nanopb #699: "has_key" still in Scons tests

ssevans87, 2021-08-20:
Thank you for all your hard work. 

I am attempting to run the unit tests but am receiving the error {AttributeError: 'SConsEnvironment' object has no attribute 'has_key':}. I am finding it at a minimum in tests\site_scons\site_tools\nanopb.py, If i am not mistaken that call is completely removed from python 3.x. 

This is my first time posting on github, so if the structure of my issue is incorrect i apologize.

ssevans87, 2021-08-20:
Upon further review, I realize that it was the Scons object that had the attribute has_key. That has been removed in the newest release of Scons(4.2) https://scons.org/scons-420-is-available.html.

PetteriAimonen, 2021-08-20:
I believe this problem has already been fixed in bdfaeaa09501f520953372

closed 2022-05-30 by PetteriAimonen