nanopb/nanopb #700: Transitive imports (i.e. public) do not work when referencing enums

mrkita, 2021-09-10:
This occurs with 0.4.3, but looking at the codebase, I presume occurs with the latest release as well.

A similar report can be found in #592.

However, I don't see a workaround while maintaining the use of the public import.

If I have a base proto file with an enum
`
//importA.proto

syntax = "proto2";

enum Test {
        TEST1 = 1;
        TEST2 = 2;
}
`

which is imported into
`
//importB.proto

syntax = "proto2";

import public "importA.proto";
`

which is imported into
`
//test.proto
syntax = "proto2";

import "importB.proto";

message Dummy {
    required Test value=1;
}
`

Then if I I run the generator
`nanopb_generator -I . test.proto`
or
`protoc --plugin=protoc-gen-nanopb=/opt/nanopb-gen/protoc-gen-nanopb --nanopb_out="-I .:." -I . test.proto`

Then I receive the following error:
Traceback (most recent call last):
  File "nanopb_generator.py", line 2151, in <module>
  File "nanopb_generator.py", line 2136, in main_plugin
  File "nanopb_generator.py", line 1984, in process_file
  File "nanopb_generator.py", line 1631, in generate_header
  File "nanopb_generator.py", line 1186, in fields_declaration
  File "nanopb_generator.py", line 1294, in default_value
KeyError: 'Test'
[33945] Failed to execute script nanopb_generator
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 255.

I understand that with a new version, I would get an exception here saying I should add all the files or use -I. However, no combination of that works when importing A. Clearly the proto parsing part from google works correctly as it doesn't complain about the enum Test not existing.

PetteriAimonen, 2021-09-10:
This is what I get with the newest git version:

<pre>
petteri@raichu:~/nanopb/test_700$ ls
importA.proto  importB.proto  test.proto

petteri@raichu:~/nanopb/test_700$ python ../generator/nanopb_generator.py test.proto
Traceback (most recent call last):
  File "../generator/nanopb_generator.py", line 1439, in default_value
    enumtype = dependencies[str(enumname)]
KeyError: 'Test'
...
Exception: Could not find enum type Test while generating default values for Dummy.
Try passing all source files to generator at once, or use -I option.

petteri@raichu:~/nanopb/test_700$ python ../generator/nanopb_generator.py importA.proto importB.proto test.proto
Writing to importA.pb.h and importA.pb.c
Writing to importB.pb.h and importB.pb.c
Writing to test.pb.h and test.pb.c

petteri@raichu:~/nanopb/test_700$ gcc -Wall -I .. -c test.pb.c

petteri@raichu:~/nanopb/test_700$ 
</pre>

So it seems to work now if all files are passed, but autodiscovery of dependent files does not seem to work in this case.

mrkita, 2021-09-10:
Interesting, for me this did not work with 0.4.3. I will try it soon with the latest to confirm.

Can this be called over the `protoc --plugin...` interface? If so how would that look?

PetteriAimonen, 2021-09-10:
Yeah, this is the equivalent through protoc:

    protoc --plugin=../generator/protoc-gen-nanopb --nanopb_out=. importA.proto importB.proto test.proto

PetteriAimonen, 2021-09-10:
A git bisect indicates that listing all files started working in commit a94b77af2b2113: Make generator consider dependencies recursively (#610)

mrkita, 2021-09-10:
Ok great thank you! I will try this today and then convince my team to update.

mrkita, 2021-09-10:
I can confirm that indeed is already fixed. Closing, and thank you again for the quick response.

closed 2021-09-10 by mrkita